scriptorron/indi_pylibcamera #1: Can't get it to work

sajmons, 2022-12-28:
Adress https://github.com/indilib/indiaspberry returns 404 error.

Your instructions:
```
 Just copy the indidevice.py and indi_pylibcamera.py in a folder.
```
Is this any folder or is it a particular folder?

I get this when running whole thing:

```
indiserver -v ./indi_pylibcamera.py
2022-12-28T15:13:33: startup: indiserver -v ./indi_pylibcamera.py
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31924 rfd=6 wfd=6 efd=7
2022-12-28T15:13:33: listening to port 7624 on fd 5
2022-12-28T15:13:33: Local server: listening on local domain at: @/tmp/indiserver
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #0
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31925 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #1
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31926 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #2
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31927 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #3
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31928 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: process 31928 exited with status 1
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #4
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31929 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #5
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31930 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #6
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31931 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: stderr EOF
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #7
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31932 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #8
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31933 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: restart #9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: pid=31935 rfd=7 wfd=7 efd=9
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: 2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T15:13:33: Driver ./indi_pylibcamera.py: Terminated after #10 restarts.
2022-12-28T15:13:33: good bye

```

scriptorron, 2022-12-28:
Hello,

it can be any folder.
You got this error:

Driver ./indi_pylibcamera.py: execlp ././indi_pylibcamera.py: Permission denied

Likely you must make the script executable:

chmod u+x indi_pylibcamera.py

Regards,

Ronald

sajmons, 2022-12-28:
After making script executable I get this:

```
stellarmate@stellarmate:~/Projects/indi_pylibcamera $ indiserver -v ./indi_pylibcamera.py
2022-12-28T22:11:14: startup: indiserver -v ./indi_pylibcamera.py
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39241 rfd=6 wfd=6 efd=7
2022-12-28T22:11:14: listening to port 7624 on fd 5
2022-12-28T22:11:14: Local server: listening on local domain at: @/tmp/indiserver
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #0
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39242 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #1
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39243 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #2
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39244 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: stderr EOF
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: process 39244 exited with status 2
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #3
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39245 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: stderr EOF
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: process 39245 exited with status 2
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #4
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39246 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #5
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39247 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: stderr EOF
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: process 39247 exited with status 2
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #6
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39248 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #7
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39249 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #8
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39250 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: ././indi_pylibcamera.py: 9: Syntax error: newline unexpected
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: restart #9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: pid=39251 rfd=7 wfd=7 efd=9
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: read: Connection reset by peer
2022-12-28T22:11:14: Driver ./indi_pylibcamera.py: Terminated after #10 restarts.
2022-12-28T22:11:14: good bye
```

scriptorron, 2022-12-29:
This is very strange. Your file "indi_pylibcamera.py" has an unexpected
newline in line 9. In the original file this line is empty. Furthermore it
is strange that this file was not executable after download. How have you
downloaded the files? You can either download as ZIP:
[image: Screenshot_20221229_112437.png]

or you can use git to make a working copy:
git clone ***@***.***:scriptorron/indi_pylibcamera.git

Git creates a subfolder with all files. The downloaded ZIP needs to be
unpacked :-)

In both cases the indi_pylibcamera.py should already be executable. And it
should not have modifications from the original.

scriptorron, 2022-12-29:
Forgot to mention:
I still did not found the time to continue on the driver. But not later
than mid January I will implement buttons to configure other raw formats
and processed frames. As soon as the sky is clear I will have autoguide
with Raspberry Pi V1 camera + PHD2 working and make pictures with HQ
camera.

sajmons, 2022-12-29:
> This is very strange. Your file "indi_pylibcamera.py" has an unexpected newline in line 9. In the original file this line is empty. Furthermore it is strange that this file was not executable after [...]

My bad. Sorry! I downloaded file with save as option (yeah I know... no comment). After making proper git clone it works!

sajmons, 2022-12-29:
> Forgot to mention: I still did not found the time to continue on the driver. But not later than mid January I will implement buttons to configure other raw formats and processed frames. As soon as [...]

Wow, that's amazing. I'm trying to achieve something simmilar with my HQ camera. You can guide with Raspberry Pi V1 camera!? That's amazing, you do OAG guiding? Would be great to get your driver displayed in Ekos UI. How you comunicate with your driver? Using indi_setprop? Do you have any examples for that?

scriptorron, 2022-12-29:
Hi,

last Autumn I made some experiments with Astroberry (on a Raspberry 4), V1
camera (connected to the same Raspberry) and PHD2. I looked with the camera
through the ocular of my telescopes finderscope. It was not easy to mound
it stable to the finderscope (see attached pictures). I also needed to buy
tele-lenses for the camera on ebay because my telescope has 2000mm focal
length and I wanted at least 200mm focal length for the finderscope+camera
construction (25mm telelenses together with 8x finderscope). After all I
got PHD2 to guide my telescope in a sufficient and stable way. But there
are still 2 issues to solve:

   1. I used the indi_rpicam driver. This one is so buggy! Setting exposure
   time is a lottery. I also have doubts that this driver reads raw images.
   For instance it makes a difference if you disabled the defect pixel
   correction algorithm (DPC) of the camera. This algorithm runs in the image
   signal processor and does not influence raw frames. Likely what I got were
   processed frames. I believe with raw frames the guiding will work much
   better.
   2. My finderscope+camera construction is not easy to focus. There are 3
   degrees of freedom: the finderscope focus, the distance between finderscope
   ocular and camera lens, the distance between camera lens and imager chip. I
   could focus on a star for guiding. But that focused area was only in the
   center of the image (or a ring around the center of the image). It seems
   that I still not found a flat image plane. I need to do spend more time in
   optimizing this. Or I need to spend money for a separate guiding scope.

I think a V1 camera is good enough for guiding. It does not allow more than
6s exposure time but for guiding you will anyway not use long exposure
times. Compared to the HQ camera the V1 has lower sensitivity, more noise
and suffers from amplifier glow. But for guiding on bright stars it is
likely good enough.

Definitely needed is a better INDI driver for the camera. My plan is to
have a HQ camera connected to a Raspberry Pi Zero for doing long exposures
through the main scope. This Pi is configured as "USB Gadget"
https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget/ethernet-gadget.
The USB connection between the Pi Zero and the Pi 4 supplies the HQ camera
with power and makes a LAN connection between the two Pi's. The Pi 4 runs
KStars, EKOS and PHD2. It also controls the mount. The V1 camera for
guiding is connected to the Pi 4 to safe the time for image transmission.
With remote desktop (xrdp) I will log in the Pi4 from a laptop to operate
everything.

You asked how I communicate with the driver. For development I start
"indiserver -v indi_pylibcamera" on the Pi Zero and connect to that with
EKOS (KStars) running on a laptop. Later EKOS will run on the Pi 4. EKOS
detects the driver (see attached screen shots, sorry - German) and I can
take photos with EKOS. But there is still a lot more to implement and
improve on the driver.

Best Regards and Happy New Year,
Ronald

sajmons, 2023-01-02:
Thnx for your help Ronald.
Now I can successfully run the driver, but when trying to connect to it with Kstars/Ekos I get this:
```
stellarmate@stellarmate:~/Projects/indi_pylibcamera $ indiserver -v ./indi_pylibcamera.py
2023-01-02T17:11:51: startup: indiserver -v ./indi_pylibcamera.py
2023-01-02T17:11:51: Driver ./indi_pylibcamera.py: pid=64287 rfd=6 wfd=6 efd=7
2023-01-02T17:11:51: listening to port 7624 on fd 5
2023-01-02T17:11:51: Local server: listening on local domain at: @/tmp/indiserver
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: [273:58:50.050524784] [64287]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: [273:58:50.127853466] [64288]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: root-INFO- found cameras: [{'Model': 'imx477', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx477@1a'}]
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: root-INFO- Parsed data from client
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: root-INFO- <getProperties version="1.7"/>
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: 
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: root-INFO- End client data
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: root-INFO- send_defVector: <defSwitchVector device="indi_pylibcamera" rule="OneOfMany" perm="rw" state="Idle" group="Main Control" label="Camera" name="CAMERA_SELECTION"><defSwitch name="CAM0" label="/base/soc/i2c0mux/i2c@1/imx477@1a">On</defSwitch></defSwitchVector>
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: root-INFO- send_defVector: <defSwitchVector device="indi_pylibcamera" rule="OneOfMany" perm="rw" state="Idle" group="Main Control" label="Connection" name="CONNECTION"><defSwitch name="CONNECT" label="Connect">Off</defSwitch><defSwitch name="DISCONNECT" label="Disonnect">On</defSwitch></defSwitchVector>
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: root-INFO- send_defVector: <defTextVector device="indi_pylibcamera" perm="ro" state="Idle" group="General Info" label="Driver Info" name="DRIVER_INFO"><defText name="DRIVER_NAME" label="Name">indi_pylibcamera</defText><defText name="DRIVER_EXEC" label="Exec">././indi_pylibcamera.py</defText><defText name="DRIVER_VERSION" label="Version">1.0.0</defText><defText name="DRIVER_INTERFACE" label="Interface">2</defText></defTextVector>
2023-01-02T17:11:52: Driver ./indi_pylibcamera.py: root-INFO- send_defVector: <defSwitchVector device="indi_pylibcamera" rule="OneOfMany" perm="rw" state="Idle" group="Options" label="Logging" name="LOGGING_LEVEL"><defSwitch name="LOGGING_DEBUG" label="Debug">Off</defSwitch><defSwitch name="LOGGING_INFO" label="Info">On</defSwitch><defSwitch name="LOGGING_WARN" label="Warning">Off</defSwitch><defSwitch name="LOGGING_ERROR" label="Error">Off</defSwitch></defSwitchVector>
2023-01-02T17:12:24: Client 9: new arrival from 127.0.0.1:37854 - welcome!
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: root-INFO- Parsed data from client
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: root-INFO- <getProperties version="1.7"/>
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: 
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: root-INFO- End client data
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: root-INFO- send_defVector: <defSwitchVector device="indi_pylibcamera" rule="OneOfMany" perm="rw" state="Idle" group="Main Control" label="Camera" name="CAMERA_SELECTION"><defSwitch name="CAM0" label="/base/soc/i2c0mux/i2c@1/imx477@1a">On</defSwitch></defSwitchVector>
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: root-INFO- send_defVector: <defSwitchVector device="indi_pylibcamera" rule="OneOfMany" perm="rw" state="Idle" group="Main Control" label="Connection" name="CONNECTION"><defSwitch name="CONNECT" label="Connect">Off</defSwitch><defSwitch name="DISCONNECT" label="Disonnect">On</defSwitch></defSwitchVector>
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: root-INFO- send_defVector: <defTextVector device="indi_pylibcamera" perm="ro" state="Idle" group="General Info" label="Driver Info" name="DRIVER_INFO"><defText name="DRIVER_NAME" label="Name">indi_pylibcamera</defText><defText name="DRIVER_EXEC" label="Exec">././indi_pylibcamera.py</defText><defText name="DRIVER_VERSION" label="Version">1.0.0</defText><defText name="DRIVER_INTERFACE" label="Interface">2</defText></defTextVector>
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: root-INFO- send_defVector: <defSwitchVector device="indi_pylibcamera" rule="OneOfMany" perm="rw" state="Idle" group="Options" label="Logging" name="LOGGING_LEVEL"><defSwitch name="LOGGING_DEBUG" label="Debug">Off</defSwitch><defSwitch name="LOGGING_INFO" label="Info">On</defSwitch><defSwitch name="LOGGING_WARN" label="Warning">Off</defSwitch><defSwitch name="LOGGING_ERROR" label="Error">Off</defSwitch></defSwitchVector>
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: root-INFO- Parsed data from client
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: root-INFO- <newSwitchVector device="indi_pylibcamera" name="CONNECTION">
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py:     <oneSwitch name="CONNECT">
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: On
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py:     </oneSwitch>
2023-01-02T17:12:24: Driver ./indi_pylibcamera.py: </newSwitchVector>
[... cut ...] Text></defTextVector>
2023-01-02T17:12:26: Driver ./indi_pylibcamera.py: root-INFO- send_defVector: <defSwitchVector device="indi_pylibcamera" rule="OneOfMany" perm="rw" state="Idle" group="Options" label="Logging" name="LOGGING_LEVEL"><defSwitch name="LOGGING_DEBUG" label="Debug">Off</defSwitch><defSwitch name="LOGGING_INFO" label="Info">On</defSwitch><defSwitch name="LOGGING_WARN" label="Warning">Off</defSwitch><defSwitch name="LOGGING_ERROR" label="Error">Off</defSwitch></defSwitchVector>
```

scriptorron, 2023-01-03:
Your camera was recognized:
`root-INFO- found cameras: [{'Model': 'imx477', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx477@1a'}]`

But errors occurred when opening the camera:
`RuntimeError: Non-owning holder (loaded_as_shared_ptr).`
`RuntimeError: Camera __init__ sequence did not complete.`

Was the camera already opened and in use by an other program?

Does the camera work with the libcamera apps? Do you get errors when you run:
`libcamera-hello --list-cameras`
`libcamera-still -r -o test.jpg`

If all this does not give a hint we need to find out the differences in our installation. Please forward me the outputs of:
`cat /etc/os-release`
`uname -a`
`apt list --installed | grep libcamera`
apt-list --installed | grep picamera`

sajmons, 2023-01-03:
I tryed again and reboted device. Same error happened again. So now I'm sure camera was not used by other device.
```
libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]

libcamera-still -r -o test.jpg
Made X/EGL preview window
[0:06:15.342289179] [2106]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[0:06:15.436879300] [2108]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:06:15.439898121] [2106]  INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
[0:06:15.442013971] [2108]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:06:21.110365388] [2106]  INFO Camera camera.cpp:1026 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
[0:06:21.112770963] [2108]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Still capture image received
Bayer format is BGGR-12
```

Both these commands gave expected result. My camera is connected to board with ribbon cable, not usb. Maybe this is waht is causing error?

scriptorron, 2023-01-03:
Sorry, I meant `apt list --installed | grep picamera`

sajmons, 2023-01-03:
```
apt list --installed | grep picamera

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

python3-picamera2/stable,stable,now 0.3.7-1 all [installed]
```

scriptorron, 2023-01-03:
There are some differences in our setups:
- We both have the same version of picamera2.
- Our versions of the libcamera packages differ. For instance your libcamera0 version is "1.0-stable\~202211161314" while my version is "0\~git20221130+0684c373-1". It seems Raspbian uses development versions while you use stable releases which are older (November 16th vs November 30th).
- Your OS is a Debian based Stellarmate while I use Raspbian. Both are Bullseye. But I have the 32bit OS while you use the 64bit one.

Do you get updates when you run `sudo apt install libcamera0 python3-libcamera`?

In worst case you found a libcamera or picamera2 bug which is only visible in 64bit OS. All tests I did so far were with 32bit Raspbian. If it is really a libcamera/picamera2 bug we can only report this to the picamera2 developers hoping to get a solution.

sajmons, 2023-01-04:
Here is output of latest command:

```
sudo apt install libcamera0 python3-libcamera
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libcamera0 is already the newest version (1.0-stable~202211161314).
libcamera0 set to manually installed.
python3-libcamera is already the newest version (1.0-stable~202211161314).
python3-libcamera set to manually installed.
The following packages were automatically installed and are no longer required:
  linux-headers-5.10.0-19-arm64 linux-headers-5.10.0-19-common
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Nothing get's updated.

scriptorron, 2023-01-04:
Okay. I will setup a 64bit Raspbian and try if I can reproduce the error. Please give me 2 days.

sajmons, 2023-01-04:
You are really great! If you can provide me with information what IDE are you using for Python development, maybe I can try to debug it myself.

sajmons, 2023-01-04:
Maybe you can see this project https://github.com/Gordon999/Pi_LIbCamera_GUI. It works on my system.

scriptorron, 2023-01-04:
Thank you for showing me this project. I will have a look on it.

I set up a Raspberry Pi 4 with 64bit Raspbian. The library versions are the same as on my 32bit Pi Zero. Unfortunately I could not reproduce the error there. In the meantime I believe you will need a library update :-(

For Python coding I use PyCharm ([https://www.jetbrains.com](https://www.jetbrains.com)) on laptop and copy the modified files with `rsync` to the Raspberry Pi. In fact I do not know how to do remote debugging. I always use `print` statements for debugging. 

To isolate the root cause I made a short and simple test case. It just detects connected cameras and opens the first one. On my systems it runs fine. On your system the line `picam2 = Picamera2(0)` will likely throw an error. 

Please copy the text below in a file "test.sh". Take care for the different indenting. Make it executable (`chmod u+x test.sh`) and run it. I expect you will get an error trace.

```python
#!/usr/bin/env python3

from picamera2 import Picamera2

# get list of available cameras:
cameras = Picamera2.global_camera_info()

if len(cameras) > 0:
    print(f'Opening camera 0: {cameras[0]}')
    picam2 = Picamera2(0)
    print('Success.')
else:
    print('WARNING: No camera found!')

print('Finished.')
```

sajmons, 2023-01-04:
The output os test.sh is:
```
stellarmate@stellarmate:~/Projects/indi_pylibcamera $ ./test.sh
[6:08:15.761089706] [8647]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[6:08:15.822850428] [8648]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media2
Opening camera 0: {'Model': 'imx477', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx477@1a'}
[6:08:15.825476481] [8647]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[6:08:15.890340713] [8652]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media2
Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 237, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 447, in _open_camera
    if not self._initialize_camera():
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 431, in _initialize_camera
    self.sensor_format = str(self.camera.generate_configuration([RAW]).at(0).pixel_format)
RuntimeError: Non-owning holder (loaded_as_shared_ptr).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/stellarmate/Projects/indi_pylibcamera/./test.sh", line 10, in <module>
    picam2 = Picamera2(0)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 244, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.
```

sajmons, 2023-01-04:
There is another INDI python driver project you might be interested in:
https://github.com/aaronwmorris/indi-allsky

scriptorron, 2023-01-04:
Thanks.

The error is what I expected. The line 431 in picamera2.py makes the trouble:
```python
self.sensor_format = str(self.camera.generate_configuration([RAW]).at(0).pixel_format)
```
In this line the method `generate_configuration` of the object `self.camera` is called. This object comes from libcamera library. For any reasons it raises the error `RuntimeError: Non-owning holder (loaded_as_shared_ptr)`. This is nothing I can fix or work around. Sorry. We can only hope that it is fixed in the next update.

sajmons, 2023-01-05:
I'm glad that you were able to pinpoint the source of a problem. Did you opened the issue on picamera2 github?
I also wish to thank you for your **outstanding** support and all the work you invested in this driver. I'll try to set up 32 bit system and test it with that.

scriptorron, 2023-01-05:
You are welcome. I am happy that someone will try my driver.

I can open an issue on picamera2. But when it comes to the point of testing bug fixes I will need your help. Too bad that I could not reproduce the error.

sajmons, 2023-01-05:
That's no problem, you can count on me for testing.

sajmons, 2023-01-05:
Hmm, that's interesting. I did a fresh instalation of Stellarmate 32bit and when I tryed to install required libraries I get this:
I guess Stellarmate repository is missing python3-picamera2?

```
stellarmate@stellarmate:~ $ sudo apt-get install python3 python3-lxml python3-astropy python3-picamera2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python3-picamera2
```

sajmons, 2023-01-05:
Unfortunatelly Stellarmate only comes in 64bit Bullseye version
I have tried it on fresh instalation (nothing changer only prerequisites for your driver installed) and after running test.sh result is the same:
```
stellarmate@stellarmate:~/Projects/indi_pylibcamera $ ./test.sh 
[1:01:18.484283460] [10540]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[1:01:18.550090496] [10541]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media1
Opening camera 0: {'Model': 'imx477', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx477@1a'}
[1:01:18.553176786] [10540]  INFO Camera camera_manager.cpp:299 libcamera v0.0.1
[1:01:18.623775182] [10545]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media1
Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 237, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 447, in _open_camera
    if not self._initialize_camera():
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 431, in _initialize_camera
    self.sensor_format = str(self.camera.generate_configuration([RAW]).at(0).pixel_format)
RuntimeError: Non-owning holder (loaded_as_shared_ptr).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/stellarmate/Projects/indi_pylibcamera/./test.sh", line 10, in <module>
    picam2 = Picamera2(0)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 244, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.
```

Maybe now is time to open new issue.

scriptorron, 2023-01-05:
I will open an issue on picamera2. Likely they will say it is fixed in the newer version. Than we can open an issue in Stellarmate and ask them to update their libraries.

scriptorron, 2023-01-05:
I opened issue [https://github.com/raspberrypi/picamera2/issues/502](https://github.com/raspberrypi/picamera2/issues/502)

scriptorron, 2023-03-21:
Has Stelarmate updated the libcamera packages and does it work now? Can we close this issue?

sajmons, 2023-03-27:
I have tested it again today, but still no luck. I will close this issue, because it's not related to your driver, and there is nothing you can do here.