scriptorron/indi_pylibcamera #13: missing gain config function for use with indi-allsky

gmke, 2023-03-10:
Mar 10 13:07:38 astroberry [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 990, in getCcdGain\n    raise Exception(\'Gain config not implemented for {0:s}, open an enhancement request\'.format(indi_exec))\n'
Mar 10 13:07:38 astroberry [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 548, in getCcdInfo\n    gain_info = self.getCcdGain()\n'
Mar 10 13:07:38 astroberry [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 536, in _initialize\n    ccd_info = self.indiclient.getCcdInfo()\n'
Mar 10 13:07:38 astroberry [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 884, in run\n    self._initialize()\n'
Mar 10 13:07:38 astroberry [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/allsky.py", line 119, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
Mar 10 13:07:38 astroberry [ERROR] MainProcess allsky.unhandled_exception() #36: Value: Gain config not implemented for ././indi_pylibcamera.py, open an enhancement request

gmke, 2023-03-10:
> Sorry, this is something that needs to be fixed in indi-allsky. I even do not know which version of indi-allsky you use. The Git project https://github.com/aaronwmorris/indi-allsky/blob/main/indi_allsky has a main and a dev branch and both do not use indi_pylibcamera. Furthermore in both the line 990 of indi.py can not raise such an exception.
> What I have seen so far the indi-allsky needs value GAIN of property CCD_GAIN. The indi_pylibcamera driver provides this. Likely it is just not handled the right way in indi-allsky. Please open an issue in the indi-allsky project.

Thank you so much for your kind analysis. I will report it to Aaron. I use the main branch of indi-allsky and had to manually start the indiserver to convince indi-allsky to use the puli camera.

CS, Markus

scriptorron, 2023-03-13:
Aaron, thank you very much. If all agree I would like to close this issue.

closed 2023-03-14 by scriptorron