scriptorron/indi_pylibcamera #17: Ekos exposure count down and guiding issue

sajmons, 2023-03-10:
@Ronald I have finnally had some time to test your driver with my HQ Camera imx477 under actual sky.
It was really plesent experiance. Best I had with my HQ Camera so far! Long exposures worked as expected. I did 60 sec without a glitch.

I starter indiserver remotelly over SSH with this command:
`indiserver -v ./indi_pylibcamera.py indi_eqmod_telescope indi_gpsd`
And then remotelly connected to it with Kstars 3.6.3 stable from Windows 11 machine.

What worked:

1. CCD Tab: Long esposures up to 60 sec with gain 22.23 (didn't test onger, but I guess it's working)
2. Align Tab: Capture & Solve worked really great! I't was really great feeling to see HQ Camera in action! Wow, thank you for that! ;)

What didn't worked:

1. When doing exposure, Ekos count down doesn't work. With 10 sec exposure progress stayed at value 10 all the time. I guess driver doesn't provide proper information back
[image]
3. Guiding doesn't seams to work either. I'm speculating that driver doesn't provide proper feedback to Ekos. Drift graph didn't update. (I must admit that it's my first time trying guiding with HQ Camera) Should try with Binning 2x2 enabled. Yesterday it was not enabled yet.
4. It is possible to set binning up to 10, but values greater than 2 give you error 2023-03-10T21:22:33 Failed to set binning.
[image]

It's really great to see such great progress with your driver! Keep up great work! Thank you!

scriptorron, 2023-03-12:
Thank you very much for your kind words. That makes me happy.

Good to read that solving works! When I tried it last time the solver had serious issues to detect stars. But I tried it with an older version of KStars/EKOS and the image FITS still had the garbage columns.

Regarding the not working things you listed:
1. The driver does not send data to EKOS for the countdown. I did not give that priority. But you are right: when you do long term exposures it is comfortable to know the remaining time. I will implement this in the next days.
2. It surprises me that guiding does not work. Do you use the internal guider or PHD2? Does the guider get image frames? Is a guiding star detected? Last time (2 weeks ago with an earlier version of the driver) I had a clear night and tested with PHD2. After setting up PHD2 for my Alt-Az mount I saw the drift graph and guiding pulses. But guiding accuracy was bad. Maybe this was because the old driver was not providing the correct binning information. Or I made something wrong when configuring PHD2. I will try that again in the next clear night.
3. The driver tells EKOS the binning value range. In thre driver control window, in focus tab and in solver tab the binning options are "1x1" and "2x2". Only in the camera tab the binning allows values above 2 (and it allows different values for horzontal and vertical which does not make sense). When I use the CCD simulator the camera tab does it right. I will try to find out what the CCD simualtor does different.

I will work on it.

scriptorron, 2023-03-15:
The latest version in the main branch has:
- exposure countdown implemented,
- allows aborting of an exposure and
- has the binning control fixed.

I could not test the guiding with the latest version. Too many clouds on the sky :-(

sajmons, 2023-03-16:
Great work!

Last time I forgot to mention another problem.

Steps to reproduce:
1. go to Guide tab in Ekos
2. press Loop button 
3. after some time press Stop button
4. go back to CCD tab
5. start another exposure
 
In previous version that didn't work. It seams, that aborting exposure that you implemented fixed that problem too! Awsome!

I will test guiding as soon as possible to report back my findings.

scriptorron, 2023-05-02:
I had a clear night to Monday and tested guiding with my HQ camera. The internal guider and PHD2 get images, can detect stars, calibrate and start guiding. So I think the camera does its work. But the guiding actions do not work with my setup (both internal and PHD2). For instance I see that RA of my mount drifts to one side. PHD2 starts to send pulses in the other direction, but RA continues to drift away. PHD2 increases the pulses in the other direction without success. And than RA of my mount "jumps" to the other side, far too much. It does not matter how I adjust the regulation parameter in PHD2, that cycling newer stops or gets lower.

This is very frustrating because I did all the effort with the indi_pylibcamera driver to get auto-guiding working (the gear imperfections limit me to 20s exposure time without auto-guiding).

Maybe the root cause is my mount driver. INDI has 2 or 3 candidates for my mount and maybe I took the wrong one. Or I need to update the mount firmware. All that is off-topic here. If you agree I will close this issue as solved.

sajmons, 2023-05-03:
What mount are you using?

scriptorron, 2023-05-04:
I tested with a Skywatcher Synscan [https://www.astroshop.de/teleskope/skywatcher-teleskop-n-150-750-star-discovery-p1-50i-synscan-wifi-goto/p,57364](https://www.astroshop.de/teleskope/skywatcher-teleskop-n-150-750-star-discovery-p1-50i-synscan-wifi-goto/p,57364). This is an Alt/Az mount and I followed instructions in [https://telescopemount.org/guiding-the-panther-alt-az-mount-with-phd2/](https://telescopemount.org/guiding-the-panther-alt-az-mount-with-phd2/).

The mount is controlled by the Synscan App on my Android phone. The INDI driver I used is [indi_synscan_telescope](https://www.indilib.org/telescopes/skywatcher/synscan-telescope.html). 

Before I connect the INDI mount driver I do alignment in the App. With the INDI driver I can control the mount in KStars (mount goes to targets with pretty good precision). Tracking also works fine. When PHD2 does calibration I can also see the guiding star moving. That makes me believe PHD2 can control the mount.

When taking 20s exposures without PHD2 autoguiding the stars are nearly round. With PHD2 autoguiding the stars are wiggled. That also makes me believe that PHD2 can control the mount.

The problem is the stability of the regulation loop. With the settings from the link above the regulation overshoots and never comes to a stable operation. Tweaking the settings did not help. It is like the mount ignores guiding pulses or it has a very long dead time. 

I will update the Synscan App to the Pro version. I will also try other INDI driver ([indi_azgti_telescope](https://www.indilib.org/telescopes/skywatcher/az-gti.html), [indi_skywatcherAltAzMount](https://www.indilib.org/individuals/devices/telescopes/skywatcher/skywatcher-alt-az-dobsonian.html)).

I also plan to test it with my [Meade LXD55](https://astromart.com/classifieds/astromart-classifieds/telescope-reflectors/show/meade-8-sc-8-f10-sct-with-standard-coatings-and-lxd55-equatorial-mount). This is an equatorial mount (PHD2 was mad for such mounts) and will have a different INDI driver. When it works with the Meade than I can eliminate the camera as error source.

Unfortunately all this needs to be done in clear nights, which are seldom this year.

By the way, yesterday I fixed a bug in the SCALE value stored in the FITS images. The wrong SCALE confused the field solver. After fixing I can solve in ASTAP and EKOS (using Load & Slew). I recommend to update your installation.

closed 2023-05-04 by sajmons