scriptorron/indi_pylibcamera #19: IMX462 not working with the driver

gordon-freimann, 2023-03-20:
Hello there,

thanks for your great work on this driver! 
You are probably my last hope to get this IMX462 camera module from arducam working with indi.
Link to Camera module:
(https://www.arducam.com/product/arducam-for-raspberry-pi-ultra-low-light-camera-1080p-hd-wide-angle-pivariety-camera-module-based-on-1-2-7inch-2mp-starvis-sensor-imx462-compatible-with-raspberry-pi-isp-and-gstreamer-plugin/) 

I managed to make the cam work. Libcamera-hello shows me a preview. 

Starting an indiserver via "indiserver -v /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py"
shows the following:
```
david@astro:~/pylibcamera/indi_pylibcamera-main $ sudo indiserver -v /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py
2023-03-20T21:00:40: startup: indiserver -v /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py2023-03-20T21:00:40: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: pid=3904 rfd=6 wfd=6 efd=7
2023-03-20T21:00:40: listening to port 7624 on fd 5
2023-03-20T21:00:40: Local server: listening on local domain at: @/tmp/indiserver
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:27:15.660482974] [3904]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4224-42eae600
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:27:15.693126882] [3905]  WARN CameraSensorProperties camera_sensor_properties.cpp:243 No static properties available for 'arducam-pivariety'
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:27:15.693186215] [3905]  WARN CameraSensorProperties camera_sensor_properties.cpp:245 Please consider updating the camera sensor properties database
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:27:15.701598838] [3905] ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'arducam-pivariety.json' not found for IPA module 'raspberrypi'
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:27:16.004228685] [3905]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/arducam_pivariety@c to Unicam device /dev/media2 and ISP device /dev/media1
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- found cameras: [{'Model': 'arducam-pivariety', 'Location': 2, 'Rotation': 0, 'Id': '/base/soc/i2c0mux/i2c@1/arducam_pivariety@c'}]
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- Parsed data from client:
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: <getProperties version="1.7"/>
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: 
2023-03-20T21:00:41: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- End client data
```
Looks fine for me.
Then I try to connect to the indi server via kstars/ekos. The console shows me the following error:

```
2023-03-20T21:03:56: Client 9: new arrival from local pid 3962 (user: 1000:1000) - welcome!
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- Parsed data from client:
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: <getProperties version="1.7"/>
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: 
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- End client data
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- Parsed data from client:
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: <newSwitchVector device="indi_pylibcamera" name="CONNECTION">
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py:     <oneSwitch name="CONNECT">
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: On
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py:     </oneSwitch>
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: </newSwitchVector>
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: 
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- End client data
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- connecting to camera /base/soc/i2c0mux/i2c@1/arducam_pivariety@c
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- closing camera
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- closing camera
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- opening camera
2023-03-20T21:03:56: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:30:31.590035201] [3956]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4224-42eae600
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:30:31.640375001] [3980]  WARN CameraSensorProperties camera_sensor_properties.cpp:243 No static properties available for 'arducam-pivariety'
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:30:31.640480426] [3980]  WARN CameraSensorProperties camera_sensor_properties.cpp:245 Please consider updating the camera sensor properties database
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:30:31.648582895] [3980] ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'arducam-pivariety.json' not found for IPA module 'raspberrypi'
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:30:31.955973871] [3980]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/arducam_pivariety@c to Unicam device /dev/media2 and ISP device /dev/media1
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: picamera2.picamera2-INFO- Initialization successful.
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: picamera2.picamera2-INFO- Camera now open.
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- camera properties: {'Model': 'arducam-pivariety', 'Location': 2, 'Rotation': 0, 'PixelArraySize': (1920, 1080), 'PixelArrayActiveAreas': (libcamera.Rectangle(0, 0, 1920, 1080),), 'ColorFilterArrangement': 0, 'ScalerCropMaximum': (0, 0, 0, 0)}
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:30:31.966919632] [3956]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-XBGR8888 (1) 1920x1080-SRGGB10_CSI2P
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: [0:30:32.026269911] [3980]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/arducam_pivariety@c - Selected sensor format: 1920x1080-SRGGB10_1X10 - Selected unicam format: 1920x1080-pRAA
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: picamera2.picamera2-INFO- Configuration successful!
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: Traceback (most recent call last):
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: Exception in thread Thread-1:
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py:   File "/home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py", line 756, in <module>
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: Traceback (most recent call last):
2023-03-20T21:03:57: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py:   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
[... cut ...]
2023-03-20T21:03:59: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: <getProperties version="1.7"/>
2023-03-20T21:03:59: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: 
2023-03-20T21:03:59: Driver /home/david/pylibcamera/indi_pylibcamera-main/indi_pylibcamera.py: root-INFO- End client data
```

Can you explain to me what this error means and how I can fix it?
I attached the MyCam.txt file your script creates. Maybe it can help.

Kind Regards,
David

[MyCam.txt](https://github.com/scriptorron/indi_pylibcamera/files/11023062/MyCam.txt)

sajmons, 2023-03-21:
@gordon-freimann can you give some information about your system?
Please run these comands in your terminal:

`cat /etc/os-release`

`uname -a`

Quick google search on your error (ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'arducam-pivariety.json' not found for IPA module 'raspberrypi').

You can try to change dtoverlay setting as suggested here:
https://forum.arducam.com/t/libcamera-error-with-pivariety-camera-imx462/2077

Or you can try to install ArduCam driver for your camera as sugested here:
https://forums.raspberrypi.com/viewtopic.php?t=317401

Or you can contact ArduCam directly on their forum: https://www.arducam.com/forum/

@scriptorron is the author of indi_pylibcamera driver, maybe he can provide further help.

sajmons, 2023-03-21:
@gordon-freimann Maybe this can help too:
https://levelup.gitconnected.com/arducam-hawk-eye-64mp-camera-module-for-the-raspberry-pi-4f451b95ed11

scriptorron, 2023-03-21:
@sajmons Many thanks for helping out.

@gordon-freimann sajmons is on the right track: the libcamera version you have does not fully support your camera. At least it does not find a configuration file 'arducam-pivariety.json' that holds information about your camera. Therefore libcamera does not provide information about UnitCellSize and it allows one raw mode only (in your MyCam.txt there is no entry for UnitCellSize in "Camera properties" and the list "Raw sensor modes" has only one entry for 1920x1080 pixel 10bit). The missing UnitCellSize was the reason for the KeyError in the indi_pylibcamera.

Likely most of the information libcamera knows about your camera are fallback-defaults and do not use all features of your camera. For instance the data sheet states that your camera also provides 12bit data. I highly recommend to fix the libcamera issue first. If that is not possible I can add a parameter in the configuration file which allows you to force a UnitCellSize. But that will still not allow you to use other raw modes or 12bit/pixel.

gordon-freimann, 2023-03-21:
Thanks for trying to help. I will check the hints you gave tomorrow and post about the results.

gordon-freimann, 2023-03-27:
I tried the following to get rid of the errors:
1. reinstalling everything, follow the arducam instructions  (https://docs.arducam.com/Raspberry-Pi-Camera/Pivariety-Camera/Quick-Start-Guide/) step by step --> I can see a preview of my cam but the UnitCellSize-Error remains
2. installing the patch suggested from here: https://forum.arducam.com/t/libcamera-error-with-pivariety-camera-imx462/2077
(since there were no installation instructions given, I replaced the provided ipa_rpi.so in /usr/lib/arm-linux-gnueabihf/libcamera)
raises the following errors when i start the indiserver:
`david@raspberrypi:~/Downloads/indi_pylibcamera-main $ indiserver -v ./indi_pylibcamera.py
2023-03-27T19:49:57: startup: indiserver -v ./indi_pylibcamera.py 
2023-03-27T19:49:57: Driver ./indi_pylibcamera.py: pid=7636 rfd=3 wfd=6 efd=7
2023-03-27T19:49:57: listening to port 7624 on fd 4
2023-03-27T19:49:58: Driver ./indi_pylibcamera.py: [1:19:19.502820775] [7636]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4225-74d023d8
2023-03-27T19:49:58: Driver ./indi_pylibcamera.py: [1:19:19.537608515] [7637]  WARN CameraSensorProperties camera_sensor_properties.cpp:243 No static properties available for 'arducam-pivariety'
2023-03-27T19:49:58: Driver ./indi_pylibcamera.py: [1:19:19.537685570] [7637]  WARN CameraSensorProperties camera_sensor_properties.cpp:245 Please consider updating the camera sensor properties database
2023-03-27T19:49:58: Driver ./indi_pylibcamera.py: [1:19:19.634081216] [7637] ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'arducam-pivariety.json' not found for IPA module 'raspberrypi'
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636449412] [7637] ERROR IPCPipe ipc_pipe_unixsocket.cpp:134 Call timeout!
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636526596] [7637] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636549633] [7637] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:190 Failed to call init
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636574781] [7637] ERROR RPI raspberrypi.cpp:1352 Failed to load a suitable IPA library
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636640613] [7637] ERROR IPCUnixSocket ipc_unixsocket.cpp:191 Failed to send: Connection refused
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636672372] [7637] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636691761] [7637] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:476 Failed to call unmapBuffers
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636824407] [7637] ERROR IPCUnixSocket ipc_unixsocket.cpp:191 Failed to send: Transport endpoint is not connected
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636853592] [7637] ERROR IPCPipe ipc_pipe_unixsocket.cpp:80 Failed to call async
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: [1:19:21.636932183] [7637] ERROR RPI raspberrypi.cpp:1277 Failed to register camera arducam-pivariety 10-000c: -22
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: root-INFO- found cameras: []
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: root-INFO- Parsed data from client:
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: <getProperties version="1.7"/>
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: 
2023-03-27T19:50:00: Driver ./indi_pylibcamera.py: root-INFO- End client data
`
but anyhow as an arducam support team member states here: 
https://forum.arducam.com/t/libcamera-produces-illegal-instruction-error/1584/7
the error: `ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'arducam-pivariety.json' not found for IPA module 'raspberrypi'`
would make libcamera read the camera properties from the device itself. 

I extensively read the arducam forums and it seems that the warnings and errors shown here:
`2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: [0:56:36.464540307] [7220]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4225-74d023d8
2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: [0:56:36.499794840] [7222]  WARN CameraSensorProperties camera_sensor_properties.cpp:243 No static properties available for 'arducam-pivariety'
2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: [0:56:36.499856691] [7222]  WARN CameraSensorProperties camera_sensor_properties.cpp:245 Please consider updating the camera sensor properties database
2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: [0:56:36.509399533] [7222] ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'arducam-pivariety.json' not found for IPA module 'raspberrypi'
2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: [0:56:36.813717850] [7222]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/arducam_pivariety@c to Unicam device /dev/media1 and ISP device /dev/media3
2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: root-INFO- found cameras: [{'Model': 'arducam-pivariety', 'Location': 2, 'Rotation': 0, 'Id': '/base/soc/i2c0mux/i2c@1/arducam_pivariety@c'}]
2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: root-INFO- Parsed data from client:
2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: <getProperties version="1.7"/>
2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: 
2023-03-27T19:27:15: Driver ./indi_pylibcamera.py: root-INFO- End client data
`
are reported from many other users. The arducam-team also encourages the users to ignore them. And telling that those errors and warnings might be fixed in the future.

@scriptorron could you maybe add the line u suggested in the config file? Maybe I can then use the camera with indi and kstars at all....

scriptorron, 2023-03-28:
I will implement settings in the config file to force a pixel size.

It seems the arducam-team takes the easy way. A normal user does not care about pixel size. So what they did is good for most of the users. But for astrophotography you need to know the pixel size and the lens focal length to calculate the viewing angle for each pixel (or the whole image). That's needed for field solver (needs to know how large the section of the sky on your image is) and auto-guider (needs to know how much the telescope position needs to be corrected to compensate the displacement of the guiding star in your image). Anyway, I will make a workaround. Please give me some days. It is not a big task but I am busy this week with other things. I will come back to you when it is finished.

scriptorron, 2023-03-30:
@gordon-freimann The main branch allows now to set pixel size in the INI file. Please:

- update the driver
- copy "indi_pylibcamera.ini" in the folder "~/.indi_pylibcamera"
- edit "~/.indi_pylibcamera/indi_pylibcamera.ini" and activate the lines
```
force_UnitCellSize_X=2900
force_UnitCellSize_Y=2900
```

I hope that will allow you to make pictures. If a new error comes up please tell me here.

gordon-freimann, 2023-03-31:
@scriptorron Thank you very much! I will try it out on sunday evening. Unfortunately I'm away from the setup for the next 2 days.

gordon-freimann, 2023-04-04:
@scriptorron It did work. Thank you!

closed 2023-04-04 by gordon-freimann