scriptorron/indi_pylibcamera #24: Property 'POOLING_PERIOD' of device 'indi_pylibcamera' not found

sajmons, 2023-04-18:
I came accross this interesting project https://indilib.org/forum/clients/13006-astralite-simple-indi-client-for-astrophotography.html
https://github.com/art-den/astra_lite. I have runed it on Windows 11 and successfully connected to INDI server. But when I tried to take picture it gives me this error. 
[image]

I'm wandering if this is indi_pylibcamera issue or AstraLite client?

scriptorron, 2023-04-20:
I asked Denis (developer of astralite) on [https://indilib.org/forum/clients/13006-astralite-simple-indi-client-for-astrophotography.html](https://indilib.org/forum/clients/13006-astralite-simple-indi-client-for-astrophotography.html) about use of POLLING_PERIOD and the new BLOB issue. It is still not sure if the incompatibility is caused by indi_pylibcamera or astralite. We will find a solution :-)

scriptorron, 2023-04-23:
Denis gave me the hint and I verified POLLING_PERIOD with the CCD Simulator: it seems to set the update rate for telling the exposure progress. The present version of indi_pylibcamera does this every 0.5 seconds fixed. I will implement POLLING_PERIOD to make this adjustable.

It is very annoying that the function of many device properties is not documented. The CCD Simulator helps to get an idea but it is still a guesswork.

closed 2023-04-24 by sajmons