scriptorron/indi_pylibcamera #28: install.sh updates

aaronwmorris, 2023-04-27:
I submitted PR #27 with some fixes and updates for install.sh

My update does not require any user input as to the location of the server script.  It is automatically detected.

Also, required system python packages are installed.

There was another issue if you have a custom install of indi in /usr/local/bin, the location of the XML file needs to be updated.  That is included, as well.

scriptorron, 2023-04-27:
Aaron,

Thank you very much! I will merge it.

I am wondering why you used `tee` instead of `cat`in this code line:
```bash
sudo tee "${INDI_SERVER_USR}/share/indi/indi_pylibcamera.xml" >/dev/null <<EOL
```
It is good as it is. I just ask to learn :-)

Regards,
Ronald

aaronwmorris, 2023-04-27:
In this case, it could probably be changed to cat.  I got into the habit of using tee to write to files with sudo, but with the multiline input, it does not matter.

This is what I mean:
```
echo "foobar" | sudo tee /path/to/file
```

scriptorron, 2023-04-27:
Thank you. I feared that there is some security issue with `cat`.

No need to change your script. I merged it already.

Thank you again for your contribution.

scriptorron, 2023-04-27:
PR #27 is merged.

closed 2023-04-27 by scriptorron