scriptorron/indi_pylibcamera #35: Camera resource not released properly

sajmons, 2023-05-11:
I'm having problem with indi_pylibcamera driver after it was not properly disconected.
It looks like that some process is still using camera. 
I tried to kill indiserver, restart indi_web_manager. Nothing helps. Only reboot solves the problem.

Here is output of libcamera-hello after indi_pylibcamera not disconected properly.

```
simon@raspberrypi:~ $ killall -9 indiserver
simon@raspberrypi:~ $ libcamera-hello -t0
Made X/EGL preview window
[23:31:26.386909776] [178147]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[23:31:26.460912891] [178148]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media0
[23:31:26.461703565] [178147]  INFO Camera camera.cpp:841 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx477@1a ***
simon@raspberrypi:~ $ libcamera-hello -t0
Made X/EGL preview window
[23:35:40.759566079] [178679]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[23:35:40.807548553] [178683]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media0
[23:35:40.807800031] [178679]  INFO Camera camera.cpp:841 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx477@1a ***
simon@raspberrypi:~ $ lsof /dev/video0
COMMAND    PID  USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
python3 173840 simon   15u   CHR  81,14      0t0  483 /dev/video0
simon@raspberrypi:~ $ lsof /dev/media0
COMMAND    PID  USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
python3 173840 simon   12uW  CHR  238,0      0t0  221 /dev/media0
simon@raspberrypi:~ $ 
```
It looks like python3 process is still using camera after drver crashed.

Is it possible to check if this is the case on driver startup and kill that process? Or maybe you could remember your own process id on startup and write it down in some file. On next startup you check that file and if it containt processid you kill it?

scriptorron, 2023-05-11:
This is likely the effect I described in the `README.md` in section "Error when restarting indiserver". The INDI server starts the driver as a sub-process and when the server crashes the sub-process does not get terminated.

There can only be one `indi_pylibcamera.py` driver running on the machine. I will implement something that kills an existing driver at startup. I just need to find out how to do this in Python (without killing itself). That will certainly not work when the older driver was started by a different user, but that is not a normal use-case.

sajmons, 2023-05-11:
That would be great! Indi web manager automatically restart driver when necessary. It would greatly improved UX and robustness if you can figure that out.

scriptorron, 2023-05-12:
I implemented the killing of an old driver in branch https://github.com/scriptorron/indi_pylibcamera/tree/35-camera-resource-not-released-properly
It worked in my tests. But I found another issue:  when you disconnect a camera and connect it again the following error comes up:
```
RuntimeError: Camera __init__ sequence did not complete.
```
It is reproducible by clicking in the driver "Connect", "Disconnect" and "Connect". This can also happen when you kill a driver which had a camera connected, start a new driver and connect the camera in the new driver. After a while it works. It seems, the libcamera stack still has a resource locked until a timeout occurs or a processing finished. I will try to find a solution.

scriptorron, 2023-05-12:
I fixed the reconnect issue in the mentioned branch. It was easier than expected. Next days I will do more tests before I merge it to the master branch.

sajmons, 2023-05-13:
That's really great to hear that you managed to solve that problem. It should really graatly improve stability and robustness of imaging. Thumbs up!

scriptorron, 2023-05-14:
Today I also added code to close camera when getting killed by SIGINT or SIGTERM. I will merge to main branch. Maybe you can give ma feedback if it works in your environment. Thanks.

sajmons, 2023-05-16:
Wow! Latest version feels amazing! Changing different INDI clients feels great. There is no more issuse if user don't disconect properly. Driver just works! You did amazing job! Driver is now really robust!

closed 2023-05-16 by sajmons

scriptorron, 2023-05-17:
Thank you for your kind words. You have a big part on it: you test it in an environment I do not have. Thank you for your help!

sajmons, 2023-05-17:
I'm really glad that I can help.