scriptorron/indi_pylibcamera #37: tags or release for OS packaging

MattBlack85, 2023-05-18:
Hi!

I am actually trying to package pylibcam for ArchLInux but I don't see a way to pull a static snapshot of the repo other than pulling always from master or using git SHAs
Would it be possible to use git tags on new releases so that I can rely on "official" releases for OS packaging?

scriptorron, 2023-05-19:
Sure, that's a good idea (sorry, that I didn't it before). In future I will set tags and make releases.

Other topic: I am wondering that you will make a package for ArchLinux. I know ArchLinux as an OS for x86 machines. But the "indi_pylibcamera" is dedicated to Raspberry Pi. Is there an ArchLinux for Raspberry Pi?

sajmons, 2023-05-19:
Yeah, there is. Here is one of them https://www.indilib.org/forum/general/12964-astroarch-yet-another-raspberry-distro-for-astrophotography-for-arm64.html.

scriptorron, 2023-05-19:
Wow! Cool!

MattBlack85, 2023-05-22:
Hi @scriptorron that's correct, I started the "AstroArch" project some time ago (that's the main topic on the indi forum I started 🙃) and I got a request to add indi_pylibcamera hence I ended up here with the tag request 😂 it's all connected

scriptorron, 2023-05-22:
Nice to meet you!

Is that now okay when I make Git releases from time to time? I am still working on the driver (adding features, fixing bugs).

MattBlack85, 2023-05-22:
absolutely!

closed 2023-05-30 by scriptorron

MattBlack85, 2023-05-30:
thanks @scriptorron much appreciated!