scriptorron/indi_pylibcamera #39: python package?

MattBlack85, 2023-06-12:
Hi @scriptorron it's me again :)

I think I got extremly close to start indi_pylibcamera

```
~/data/repos/indi_pylibcamera (main*) » indiserver -vvvv /usr/bin/indi_pylibcamera                                                                                                                                                               matt@arcturus
2023-06-12T14:23:53: startup: indiserver -vvvv /usr/bin/indi_pylibcamera
2023-06-12T14:23:53: Driver /usr/bin/indi_pylibcamera: pid=772692 rfd=6 wfd=6 efd=7
2023-06-12T14:23:53: listening to port 7624 on fd 5
2023-06-12T14:23:53: Local server: listening on local domain at: @/tmp/indiserver
2023-06-12T14:23:53: Driver /usr/bin/indi_pylibcamera: sending msg nq 1:
<getProperties version="1.7"/>

2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: root-INFO- my PID: 772692
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: root-INFO- my file name: indi_pylibcamera
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: [78:39:32.969549545] [772692]  INFO Camera camera_manager.cpp:298 libcamera v0.0.5
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: [78:39:32.971149369] [772696]  WARN V4L2 v4l2_pixelformat.cpp:328 Unsupported V4L2 pixel format H264
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: root-INFO- found cameras: [{'Model': 'Razer Kiyo', 'Location': 2, 'Id': '\\_SB_.PCI0.GP13.XHC0.RHUB.PRT1-1:1.0-1532:0e03'}]
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: root-INFO- selected logging level: LOGGING_INFO
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defNumberVector indi_pylibcamera POLLING_PERIOD Idle rw
  PERIOD_MS='1000'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defNumberVector indi_pylibcamera POLLING_PERIOD Idle rw
  PERIOD_MS='1000'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defTextVector indi_pylibcamera ACTIVE_DEVICES Idle rw
 ACTIVE_TELESCOPE=''
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defTextVector indi_pylibcamera ACTIVE_DEVICES Idle rw
 ACTIVE_TELESCOPE=''
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defSwitchVector indi_pylibcamera CAMERA_SELECTION Idle rw
       CAM0='On'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defSwitchVector indi_pylibcamera CAMERA_SELECTION Idle rw
       CAM0='On'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defSwitchVector indi_pylibcamera CONNECTION Idle rw
    CONNECT='Off'
 DISCONNECT='On'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defSwitchVector indi_pylibcamera CONNECTION Idle rw
    CONNECT='Off'
 DISCONNECT='On'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defTextVector indi_pylibcamera DRIVER_INFO Idle ro
 DRIVER_NAME='indi_pylibcamera'
 DRIVER_EXEC='/usr/bin/indi_pylibcamera'
 DRIVER_VERSION='1.9.0'
 DRIVER_INTERFACE='2'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defTextVector indi_pylibcamera DRIVER_INFO Idle ro
 DRIVER_NAME='indi_pylibcamera'
 DRIVER_EXEC='/usr/bin/indi_pylibcamera'
 DRIVER_VERSION='1.9.0'
 DRIVER_INTERFACE='2'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defSwitchVector indi_pylibcamera LOGGING_LEVEL Idle rw
 LOGGING_DEBUG='Off'
 LOGGING_INFO='On'
 LOGGING_WARN='Off'
 LOGGING_ERROR='Off'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defSwitchVector indi_pylibcamera LOGGING_LEVEL Idle rw
 LOGGING_DEBUG='Off'
 LOGGING_INFO='On'
 LOGGING_WARN='Off'
 LOGGING_ERROR='Off'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defNumberVector indi_pylibcamera POLLING_PERIOD Idle rw
  PERIOD_MS='1000'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defNumberVector indi_pylibcamera POLLING_PERIOD Idle rw
  PERIOD_MS='1000'
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defTextVector indi_pylibcamera ACTIVE_DEVICES Idle rw
 ACTIVE_TELESCOPE=''
2023-06-12T14:23:54: Driver /usr/bin/indi_pylibcamera: read defTextVector indi_pylibcamera ACTIVE_DEVICES Idle rw
 ACTIVE_TELESCOPE=''
```

When I was packaging everything I realized that the actual way to install indi_pylibcamera is to clone the repo and symlinking the entrypoint to the one within the repo.

would you consider (yes, I know it's another request from me :joy: ) to rework the repo in such a way that this will become a full featured python package? This way the helpers may live under `/usr/lib/pythonX.XX/sites-packages/indi_pylibcamera` and /usr/bin/indi_pylibcamera` can be installed under /usr/bin without having to symlink

PS: it seems also that even if the xml spec is there, I don't see the driver on Kstars, any idea on this?

sajmons, 2023-06-13:
@MattBlack85 I think actually command to verbose logging of indi server is with three vvv not four :)

`indi_server -vvv`

In KStars you can find indi_pylibcamera driver inside Others section:

[image]

At least on my system wich is Raspberry PI OS Bullseye.

About rewriting repro Ronald will have to ansver. Maybe you can give an example structure of Python package?

MattBlack85, 2023-06-13:
course you right with the triple v and I see now the pylibcamera under the `Others` thanks!

to @scriptorron 

the scaffold would be more or less the following:

```
repo root
|________indi_pylibcamera
|                |
|                |__ CameraControl.py
|                |__ SnoopingManager.py
|                |__ indidevice.py
|                |__ print_camera_info.py
|
|________bin
|                |
|                |__ indi_pylibcamera
|
| setup.py
```

I packaged some python stuff in the past and if you guys like, I can try to make those changes.

This will allow basically to install the scaffold as a python package (pip, pipenv, blabla) and the executable can be installed via pip or packaged for any OS

MattBlack85, 2023-06-13:
that would be the best option but I believe it won't make it :|

from what I observed while trying to build it it has a lots of dependencies on external python libraries (which is a first in indi I believe) plus the project structure still implies you are importing a python lib.

Maybe the code from the different files could be merged into a single one (although from a maintenance perspective it is not a good thing), at that point listing the deps would be enough (although, again, I dunno how cmake can be instructed to find for python dependencies when building indi_pylib_camera

scriptorron, 2023-06-21:
It is now on PyPi (https://pypi.org/project/indi-pylibcamera/). 

Installation procedure is:
```
sudo apt-get install libcamera-apps indi-bin python3-picamera2 python3-lxml python3-astropy
sudo pip3 install indi_pylibcamera
sudo indi_pylibcamera_postinstall
```

It is important to install `python3-picamera2` with apt-get! Pip could also install it but it is very likely that `picamera2` will not fit to the system `libcamera`.

scriptorron, 2023-06-21:
I tested it on 3 Raspberries. But would be nice if someone else could test the installation.

closed 2023-06-21 by scriptorron

sajmons, 2023-06-21:
Great work @scriptorron! I'll test new installation procedure as soon as I can.

MattBlack85, 2023-06-22:
@scriptorron it works for me, I packaged also the camera and all deps for ArchLinux and it's available now as a package for my distro, many thanks!