scriptorron/indi_pylibcamera #56: Installation unter Debian bookworm

SigvaldS, 2023-12-18:
Seit neuestem läßt sich unter bookworm auf dem Raspberry Pi nicht mehr pip auf externe quellen nutzen.
Der Versuch über ein eigenes Enviroment ist leider missglückt.

Gab es nicht eine Möglichkeit den Sourcecode direkt zu nutzen?
SigvaldS

scriptorron, 2023-12-18:
Das Package is auf PyPi und kann ohne Umstände mit `pip` installiert werden. Habe gestern erst auf einem Pi5 mit Bookworm installiert. Ich installiere immer in einem Environment weil ich ganz oft Versionen wechseln muss. Dabei installiere ich immer picamera2 und numpy mit `apt`, weil sonst Abhängigkeiten mit System-Libraries nicht aufgelöst werden. Meine Prozedur ist:
- sudo apt install libcamera-apps python3-picamera2 python3-lxml python3-astropy python3-numpy python3-venv
- python3 -m venv --system-site-packages /home/cam/venv_indi_pylibcamera
- source venv_indi_pylibcamera/bin/activate
- pip install --upgrade pip
- pip install indi_pylibcamera

Aber auch Environment müßte `pip install indi_pylibcamera` funktionieren.

scriptorron, 2023-12-18:
Ach, nochwas. Auf Pi5 gibt es momentan Probleme mit den Bilddaten: picamera2 macht auf Pi5 eine andere Skalierung der Rohdaten. Siehe Issue #54. Ich arbeite an einem Fix in Branch [54-noisy-image-with-raspberry-pi-5](/scriptorron/indi_pylibcamera/tree/54-noisy-image-with-raspberry-pi-5).

scriptorron, 2024-01-12:
Hat sich das Problem gelöst?

ettoreribaudo, 2024-01-29:
Hello Roland all perfect but i have other raspberry with Raspbian 12 and your procedure have problems in :

source venv_indi_pylibcamera/bin/activate (permissione denied)
pip install --upgrade pip (error: externally-managed-environment

This environment is externally managed

To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.)

pip install indi_pylibcamera (pip install indi_pylibcamera
error: externally-managed-environment

This environment is externally managed

To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.)

Reagrds.
Ettore

scriptorron, 2024-02-28:
No response since 2 months. Seems the issue is solved.

closed 2024-02-28 by scriptorron