scriptorron/indi_pylibcamera #61: Exception with imx296 global shutter

aaronwmorris, 2024-01-30:
I receive the following exception when trying to use the IMX296 global shutter module.

I assume that because the IMX296 is monochrome, it does not have the INDI_RAW control.

Sorry for the formatting.
```
indi.newMessage() [344]: new Message 2024-01-30T16:29:44: [ERROR] Uncaught exception!&#10;Traceback (most recent call last):&#10;  File "/home/aaron/git/indi-allsky/virtualenv/indi-allsky/bin/indi_pylibcamera", line 8, in <module>&#10;    sys.exit(main())&#10;             ^^^^^^&#10;  File "/home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/indi_pylibcamera/indi_pylibcamera.py", line 1372, in main&#10;    device.run()&#10;  File "/home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/indi_pylibcamera/indidevice.py", line 898, in run&#10;    self.message_loop()&#10;  File "/home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/indi_pylibcamera/indidevice.py", line 844, in message_loop&#10;    vector.set_byClient(values)&#10;  File "/home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/indi_pylibcamera/indi_pylibcamera.py", line 126, in set_byClient&#10;    if self.parent.openCamera():&#10;       ^^^^^^^^^^^^^^^^^^^^^^^^&#10;  File "/home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/indi_pylibcamera/indi_pylibcamera.py", line 1073, in openCamera&#10;    self.knownVectors["RAW_FORMAT"].update_Binning()  # set binning according to frame type and raw format&#10;    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^&#10;  File "/home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/indi_pylibcamera/indi_pylibcamera.py", line 203, in update_Binning&#10;    if self.parent.knownVectors["CCD_CAPTURE_FORMAT"]["INDI_RAW"].value == ISwitchState.ON:&#10;       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^&#10;  File "/home/aaron/git/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/indi_pylibcamera/indidevice.py", line 220, in __getitem__&#10;    raise KeyError(f"{name} not in {self.__str__()}")&#10;KeyError: 'INDI_RAW not in <Vector SwitchVector name=CCD_CAPTURE_FORMAT, device=indi_pylibcamera>'
```

scriptorron, 2024-01-30:
Hi Aaron,

Do you use it in indi_allsky? Do you know if the driver works there with other camera types?

When I understand the log right the exception came after opening the camera. Were there other messages (warnings) before the lines you sent?

Regards,
Ronald

aaronwmorris, 2024-01-30:
I was just testing one of my lab environments and I the imx296 was the first camera I grabbed.

These the the only other lines of output of note:

```
024-01-30 10:43:57,525 [INFO] Capture-1-6071/Dummy-1 indi.newMessage() [344]: new Message 2024-01-30T16:43:57: [INFO] camera properties: {'Model': 'imx296', 'UnitCellSize': (3450, 3450), 'Location': 2, 'Rotation': 180, 'PixelArraySize': (1456, 1088), 'PixelArrayActiveAreas': [(0, 0, 1456, 1088)], 'ColorFilterArrangement': 5, 'ScalerCropMaximum': (0, 0, 0, 0), 'SystemDevices': (20749, 20737, 20738, 20739)} 
2024-01-30 10:43:57,563 [INFO] Capture-1-6071/Dummy-1 indi.newMessage() [344]: new Message 2024-01-30T16:43:57: [WARNING] raw mode not supported: {'format': R10_CSI2P, 'unpacked': 'R10', 'bit_depth': 10, 'size': (1456, 1088), 'fps': 60.38, 'crop_limits': (0, 0, 1456, 1088), 'exposure_limits': (29, None)} 
RAW_FORMAT: defSwitchVector with no valid members
```

aaronwmorris, 2024-01-30:
It looks like `COLOR_FILTER_ARRANGEMENT` value of 5 indicates a monochrome sensor.

https://libcamera.org/api-html/namespacelibcamera_1_1properties_1_1draft.html

scriptorron, 2024-01-31:
You are right with your first guess: the driver can not handle monochrome cameras. When opening a camera the driver looks for raw modes. In the present version it supports unpacked Bayer pattern raw modes only (for instance "SRGGB12"). Your camera is monochrome and its only one raw mode is "R10".

It is definitely needed to support monochrome cameras too. I will work on this. Unfortunately I do not have that camera.

Can you do me a favor, execute
```
indi_pylibcamera_print_camera_information > Raspi_GlobalShutter_IMX296.txt
```
and send me the ` Raspi_GlobalShutter_IMX296.txt` file? Thanks!

aaronwmorris, 2024-01-31:
Here you go.

[Raspi_GlobalShutter_IMX296.txt](https://github.com/scriptorron/indi_pylibcamera/files/14112953/Raspi_GlobalShutter_IMX296.txt)

scriptorron, 2024-02-28:
Hi Aaron,

I pushed a bug fix in the main branch. It implements my "best guess" for the monochrome format. Maybe you can clone the main branch and test it. If this is too complicated for you please wait until the next release. I will finish 2 other issues before I make the release. 

In case the captured frames are wrong (errors, garbage images or less gray levels than expected) please:
- enable `log_FrameInformation=yes` in the INI file,
- make a partly overexposed picture and
- send me the FITS picture and the log outputs

Regards,
Ronald

aaronwmorris, 2024-02-29:
Sounds good.  I may try this in the next few days if I get a few minutes.

closed 2024-05-26 by scriptorron