scriptorron/indi_pylibcamera #65: FITS do not have raw images. Only weird mono

profhuster, 2024-03-07:
From INDI forum:
I am fairly new to INDI stuff.
System:

- RPi Zero 2W
- RPi HQ Camera mounted at backfocus of 16"LX600
- bullseye arm64 lite. Update and upgrade
- apt install -y libcamera-apps indi-bin python3-picamera2 python3-lxml python3-astropy python3-pi
- pip3 install indi_pylibcamera
- indi_pylibcamera_postinstall
- indiserver -v indi_pylibcamera

The INDI driver started ok. Connected CCDCiel from my laptop. Connected to RPi INDI and camera indi_pylibcamera.
I was able to loop th epreview and manually focus and center the start (Procyon.) The preview was mono, of course. Then I started a sequence of 10 images. They were all uploaded to my laptop as FITS files. These images were all binned 2x2.
The next day I was looking at them and they only have a mono images and they have odd gaps every other row and column. I probed the FITS files and it only has that one image.
Did I set something up wrong in the INDI driver? If I capture using libcamera I get color images on the RPi I can move to my laptop, but the FITS images are weird.

I can't attach the FITS file, but I attached the image. There is only one item in the FITS file. Here is the file info
```
Filename: 20240304/20240305_032943782_Procyon.fits
No.    Name      Ver    Type      Cards   Dimensions   Format
  0  PRIMARY       1 PrimaryHDU      53   (2024, 1520)   int16 (rescales to uint16) 
```

[image]

profhuster, 2024-03-07:
I did not know the image had to be debayered.

closed 2024-03-07 by profhuster

scriptorron, 2024-03-08:
Exactly. The Bayer color filters are the reason for the checkerboard in the lighter regions of your image:
[image]

The header of the FITS images has an attribute "BAYERPAT" which tells the color order in the Bayer pattern. A monochrome FITS image does not has this attribute. As far as I know this makes the difference between monochrome and raw Bayer images in FITS format.

KStars has a setting to enable debayering before showing FITS images on the screen. That allows you to see colored images in the preview window. It does not affect the stored image data. Likely CCDCiel has a similar feature.

Using raw Bayer images is preferred over RGB images. All color cameras I know use Bayer filter. The RBG images are generated in an image signal processor which does the debayering. But this image signal processor usually calculates with integer numbers which leads to loss of information. This is not critical for good exposed images. But in astrophotography the interesting parts of the images are extremely underexposes: the faint nebulas between bright stars are often hidden by the camera noise. Only postprocessing makes these details visible:
- Correction of pixel errors: You should use Dark frames to subtract individual pixel bias. If you are a professional you can also use Flat frames to compensate for individual pixel sensitivities. I do not have the equipment for doing Flat frames. But the HQ camera has very homogeneous pixel sensitivity and my optics do not produce vignetting. So I don't use Flat images.
- Debayering: The best debayer algorithms are nonlinear. That's the reason why you need to do debayering after the correction of linear pixel errors.  
- Aligning and stacking: This improves your noisy pixel data: the mean grows with the number of frames while the standard deviation grows with the square root of the number of frames. The stacking of hundreds of frames makes details visible which were hidden by noise in the single frames.
- Postprocessing: Here you correct colors/contrast and apply filter.

All these steps should be done with floating point arithmetic! There is a lot of good astro software available for doing this. My preferred free software for deep-sky images is [Siril](https://siril.org/).

The 2x binning of the RPi HQ camera is aware of the Bayer pattern. It only bins pixel of the same Bayer color. The resulting is still a raw Bayer color image.

I will send results of this issue discussion to the INDI lib forum to avoid leaving a bad mark on the `indi_pylibcamera` driver. I have the impression that they promote the `indi_libcamera` driver only. I am not up-to-date with this driver but for a long time it was very unstable and did not support raw Bayer images. The `indi_pylibcamera` driver runs in my setup (2 telescopes with 3 cameras: 2x RPi3 and one RPi0) since over a year without any issues. Despite of bad seeing conditions (very bad light pollution due to a near town) I am proud of the pictures I make:
[image]
(M27, stack of 68 mages each 20 seconds)

profhuster, 2024-03-08:
Ahhhh! Thank you so much!