scriptorron/indi_pylibcamera #66: Bookworm install procedure, raspberry pi5

BigALTelescope, 2024-03-08:
raspberrypi5, bookworm install  I installed using the guides using venv, including the user Sigvalds.  My error when running indiserver is 

Driver indi_pylibcamera: execlp : No such file or directory

I know the server is running because indi_lx200classic runs.  

It seems like a simple problem.

I do not see indi diirectory in /usr/lib/python3.11/site_packages  which is not there it is dist_packages?

I have set this up in bullseye last year no problem with HQ camera.  I appreciate your work to make this available at all.

scriptorron, 2024-03-10:
Hi BigALTelescope,

The error `Driver indi_pylibcamera: execlp : No such file or directory` comes when the `indiserver` does not find the `indi_pylibcamera` executable. As far as I understand you installed the driver in a virtual environment. You need to activate this environment every time before you use it: `source venv_indi_pylibcamera/bin/activate` (or replace venv_indi_pylibcamera with the name of your virtual environment). After that the driver executable is available in that and only that shell.
To make things easy in the night I made a shell script on my Raspberry Pi:
1. Create the `bin` subfolder in your home directory: 
    ```
    cd
    mkdir bin
    ```
    This subfolder gets added to the executable search path automatically.
2. Use a text editor to create a new file `start_INDI.sh` in the subfolder `bin` with the following contents:
    ```
    #!/usr/bin/env bash
    source venv_indi_pylibcamera/bin/activate
    indiserver -v indi_pylibcamera indi_lx200autostar > ~/indiserver.log 2>&1 &
    ``` 
    This script gets executed in the `bash` shell (line 1), activates the virtual environment (line 2, if needed replace the environment name with yours) and starts the INDI server with the camera and mount driver. Replace the mount driver with the one fitting to your mount. The outputs of the INDI server are written to `indiserver.log` in your home directory.

    If you like you can also start KStars by adding a 4th line containing `kstarts`.
3. Make the script executable with
    ```
    chmod u+x ~/bin/start_INDI.sh
    ```
4. If you like make a desktop shortcut to this shell script to start it.

My EKOS profile looks like this:
[image]
Important is to set Mode to "remote" and Host to "localhost". When you click on the triangle right to the Mount input you get a list of mounts. Take the one fitting to your mount and driver. For the Camera 1 input click in the input field and write "pylibcamera Main". Save that profile with a meaningful name.

For an observation session run `start_INDI.sh` and use the profile you created.

Best Regards,
Ronald

scriptorron, 2024-05-03:
Is the issue solved?

closed 2024-05-15 by scriptorron