scriptorron/indi_pylibcamera #67: Errors in Astra_lite logs when using indi_pylibcamera

sajmons, 2024-05-05:
When using indi_pylibcamera driver 2.6.0 with Astra_lite, it kinda work, but it's unstable. I'm guessing that folowing errors have something to do with that.

I'm not quite sure if this is astra_lite related error or indi_pylibcamera, maybe it is both.

[2024-05-05 18:26:23.547905 +02:00] INFO astra_lite x86_64 ver. 0.0.90 is started
[2024-05-05 18:26:36.904190 +02:00] ERROR indi_api: 'Property `TIME_UTC` of device `GPSD` not found' for XML

<setTextVector device="GPSD" name="TIME_UTC" state="Busy" timeout="60" timestamp="2024-05-05T16:26:38">
    <oneText name="UTC"/>
    <oneText name="OFFSET"/>
</setTextVector>
[2024-05-05 18:26:36.904626 +02:00] ERROR indi_api: 'Property `GEOGRAPHIC_COORD` of device `GPSD` not found' for XML

<setNumberVector device="GPSD" name="GEOGRAPHIC_COORD" state="Busy" timeout="60" timestamp="2024-05-05T16:26:38">
    <oneNumber name="LAT">
0
    </oneNumber>
    <oneNumber name="LONG">
0
    </oneNumber>
    <oneNumber name="ELEV">
0
    </oneNumber>
</setNumberVector>
[2024-05-05 18:26:36.905167 +02:00] ERROR indi_api: 'Property `GPS_STATUS` of device `GPSD` not found' for XML

<setTextVector device="GPSD" name="GPS_STATUS" state="Busy" timeout="60" timestamp="2024-05-05T16:26:38">
    <oneText name="GPS_FIX"/>
</setTextVector>
[2024-05-05 18:26:36.905345 +02:00] ERROR indi_api: 'Property `POLARIS` of device `GPSD` not found' for XML

<setNumberVector device="GPSD" name="POLARIS" state="Busy" timeout="60" timestamp="2024-05-05T16:26:38">
    <oneNumber name="HA">
0
    </oneNumber>
</setNumberVector>
[2024-05-05 18:26:36.905481 +02:00] ERROR indi_api: 'Property `GPS_REFRESH` of device `GPSD` not found' for XML

<setSwitchVector device="GPSD" name="GPS_REFRESH" state="Busy" timeout="0" timestamp="2024-05-05T16:26:38">
    <oneSwitch name="REFRESH">
Off
    </oneSwitch>
</setSwitchVector>
[2024-05-05 18:26:36.905607 +02:00] ERROR indi_api: 'Property `GPS_STATUS` of device `GPSD` not found' for XML

<setTextVector device="GPSD" name="GPS_STATUS" state="Busy" timeout="60" timestamp="2024-05-05T16:26:38">
    <oneText name="GPS_FIX">
NO FIX
    </oneText>
</setTextVector>
[2024-05-05 18:26:46.768334 +02:00] ERROR indi_api: Malformed XML. 1:1 Unexpected token: </
[2024-05-05 18:26:47.674317 +02:00] ERROR indi_api: 'Malformed XML. 1:1 Unexpected characters outside the root element: C' for XML
CAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgIJAAkCCPoJAAj4CPoI/wj9CQAI9Qj7CPwJBAkACPUI9wj+CQAI/AkICP4JAQkLCPkJCgj3CQoJAwkACPcI/Qj3CQkI/gkHCPwI9Aj4CPcI/QjxCQAJBwkHCQEI9Aj/CQQI/QkJCQUJAwj9CPkJAgj5CQgJBAkDCQYJBgj/CPwI/Qj8CPYI/Qj7CP0JBwj9CQEI/AkACP8I/wj8CP4JBAj3CPwI9wj2CP4JBwj9CQUI9wkACQII/gj8CQEI/Aj/CP0I+gj8CPwI/wkACPkJAAj7CQUI+wj9CP4I/wj+CPwI/wkBCP0JAgj7CP4I9wkCCPcI+gj7CPwJAwj/CPoI9gj+CPIJAAj3CPsI+Aj+CQ0I/wkHCPUJAAkCCQ0JAgkBCP4JAQj/CPYI+wkFCP8JCAkMCPoJCAkFCPwJAAkACPkJBgj1CPwJCwkBCQAJAgj0CQEJBgkKCQkI/gkGCPwJAQkACPkI/wj3CP4JCAkECQIJCAkCCP8I9AkACPMI+gkFCQ0JBAj+CPwJAgj+CQII9Aj2CPYI/QkGCQIJBAj8CPsJCgjzCPkI+gj6CPII+gkACPsJAwj9CP8I+wj8CPoI/gj9CP0JAAkACQQI+wkDCQEI9gkKCPAI/wkHCQAI/Qj/CP0I+gkCCQAJBAkDCQMJCgkHCQMI+QkGCPkJAwkCCQcJBwj6CQMI9AkECPYI+gkCCPEJAAjyCPsJAgj6CPMI+QjzCPwJAAj+CQMJAQkBCP0JCQj+CQcI/gkECQEJBAj+CQkJAwj9CQQI9Aj9CQ0I/wj5CQQJAQkGCP4JBAj+CPoI+wkCCPYI+QkACP8JAAj8CPQJAQj9CPgJAgj+CQII/wkCCQsJDgj/CP8I/wkGCQgI+Aj2CQEJCQkCCPsI+AjxCPsI9gj/CQEI/QkDCQII/wkLCPkI/AkFCQEJAwkLCQYI+AkACQMJBwj+CQIJAQj+CP0JCAj6CQkI+wkCCPkI/QkCCPsI8gj9CPMJBwj8CPUI+Aj7CP8I+gj8CQAJAgj6CQQJAAkBCPoI+Qj7CQQI+QkHCQUJAQj6CP8I+QkHCPgI+Qj3CP4JAAjyCPkI/gj8CQEI9gjxCPsJCgj8CQIJBgj5CQQI/gj5CPwI+wkFCQ4JBQkMCQQI/gj+CQAJAwkFCP4I8wkDCQAJAAkKCQcJAQj7CPcI+gj8CPwJAQkECQAJAwkBCQEJAAkBCP4JBwj8CQEI+Aj6CQAJBAj8CQUJBwj6CP0JAgkACQMI+AkECPoI/QkGCPUJBgj5CPkJDAj9CQMI+wj2CPgI9AjzCPUI9QkBCPkI+QkICPkJCgj5CQUI+QkACQAJBAkQCP8I/Aj+CQII/QkFCP0JAQkDCQAJBQj5CQII/QkACQsI/QkLCP8I/AkAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
    </oneBLOB>
</setBLOBVector>
<setNumberVector device="indi_pylibcamera" name="CCD_EXPOSURE" state="Ok" timeout="60">
    <oneNumber name="CCD_EXPOSURE_VALUE">
0
    </oneNumber>
</setNumberVector>
[2024-05-05 18:30:11.306240 +02:00] ERROR indi_api: 'Property `CAMCTRL_BRIGHTNESS` of device `indi_pylibcamera` not found' for XML

<delProperty device="indi_pylibcamera" name="CAMCTRL_BRIGHTNESS"/>
[2024-05-05 18:30:11.306424 +02:00] ERROR indi_api: 'Property `CAMCTRL_COLOURGAINS` of device `indi_pylibcamera` not found' for XML

<delProperty device="indi_pylibcamera" name="CAMCTRL_COLOURGAINS"/>
[2024-05-05 18:30:11.306540 +02:00] ERROR indi_api: 'Property `CAMCTRL_CONTRAST` of device `indi_pylibcamera` not found' for XML

<delProperty device="indi_pylibcamera" name="CAMCTRL_CONTRAST"/>
[2024-05-05 18:30:11.306651 +02:00] ERROR indi_api: 'Property `CAMCTRL_EXPOSUREVALUE` of device `indi_pylibcamera` not found' for XML

<delProperty device="indi_pylibcamera" name="CAMCTRL_EXPOSUREVALUE"/>
[2024-05-05 18:30:11.306854 +02:00] ERROR indi_api: 'Property `CAMCTRL_SATURATION` of device `indi_pylibcamera` not found' for XML

<delProperty device="indi_pylibcamera" name="CAMCTRL_SATURATION"/>
[2024-05-05 18:30:11.306963 +02:00] ERROR indi_api: 'Property `CAMCTRL_SHARPNESS` of device `indi_pylibcamera` not found' for XML

<delProperty device="indi_pylibcamera" name="CAMCTRL_SHARPNESS"/>
[2024-05-05 18:30:59.678347 +02:00] ERROR indi_api: Malformed XML. 1:1 Unexpected token: </
[2024-05-05 18:31:00.600238 +02:00] ERROR indi_api: 'Malformed XML. 1:1 Unexpected characters outside the root element: I' for XML
I+wkICQsJAAj+CPMI/QkCCQQJDQj/CQ4I+Aj5CP4JAgkICQ0I/QkGCP8I/gkPCQAI/AkECP8I/AkDCQkJAgkECPwI+wkDCPoI/wkECQMJAQkACQAJAwkACP4JAgkJCPwJBgkFCP4JBAkFCQEJEQkECQYJDAkICQEJAgj9CQ4I+Qj4CQAJCQj1CQsJAAkACQQJAgkDCQAJAwj9CQEJCAj8CPsI9AkGCPcI+wkACP0I/gjyCP4JDgj5CPkJBgkFCQoJBgkACP4JCQj8CQwJCgkFCQsI/Qj8CP4JCgj6CP8I+Aj+CPwJAwj6CQEJCAj1CQUI9wkHCPYJAgkNCQAI/gkBCPwI/gj/CPII+wj2CQMJAQkGCQAJAwkCCQ4I/Qj9CQoI/Aj+CQIJBQj+CQUJAgkKCQsI/wkHCPkJBwj5CQAJBQkBCQQJAAkACPsJAwj3CQEI9wj4CPgJCQj7CPoI/gkACQYJAQkBCQYJAQkFCQYI+Qj/CPUJBQj1CQwI9wkECPkJBwjyCQEJBQkDCQAJBgj7CPgJDgj7CPgI+wj2CP4I9wkFCP0JAQkMCQEI/gkACQMI+wj+CP4JAgkCCPEJBQjvCQoJCQj3CQYI9wj2CQ8I+Aj+CQcI/QkECPoI/Qj8CQMI/AkFCQYJCAj [... cut ...] AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
    </oneBLOB>
</setBLOBVector>
<setNumberVector device="indi_pylibcamera" name="CCD_EXPOSURE" state="Ok" timeout="60">
    <oneNumber name="CCD_EXPOSURE_VALUE">
0
    </oneNumber>
</setNumberVector>

scriptorron, 2024-05-05:
Hi. The log starts with errors from a GPS device "GPSD". Do you still get errors when you start the indi_server without GPSD? Which version of Astra_lite do you use and where did you downloaded it? I will try to reproduce the error.

sajmons, 2024-05-06:
@scriptorron I think I was testing with 0.0.90 but I'm not quite sure. I build it myself. I will send you binaries when I get back home. I have opened issue on astra_lite to to provide prebuild binaries, that would make testing a lot easier.

I'm not quite sure where the issue is. In astra_lite not properly handling some XML parameters or in indi_pylibcamera.

scriptorron, 2024-05-06:
Would be very nice to get the binary.

Your log shows 2 types of errors:
```
ERROR indi_api: 'Property `****` of device `****` not found' for XML
```
The device is `GPSD` or `indi_pylibcamera`. At least for `indi_pylibcamera` the properties are non-standard. The INDI protocol allows non-standard properties and a client must ignore them if not supported. Maybe Astra_Lite gets confused by them. Or maybe that error has nothing to do with the issue.

The second type of error is critical:
```
ERROR indi_api: Malformed XML. 1:1 Unexpected token: </
ERROR indi_api: 'Malformed XML. 1:1 Unexpected characters outside the root element: C' for XML
```
The "unexpected characters outside the root element" are image data from `indi_pylibcamera`. It seems Astra_Lite received the "</" XML token between image data. This is strange, because the image data is base64 encoded and should only contain the characters `A-Z`, `a-z`, `0-9`, `+`, `/` and `=`. I can not imagine a situation where the image data stream contains "</". 

Both errors are reported from `indi_api`. This must be something in Astra_Lite. Maybe the `indi_api` receives a `GPSD` message within the `indi_pylibcamera` image data. Normally the `indi_server` serializes messages from different devices and avoids an intermix. Does Astra_Lite use the `indi_server` or has it an own replacement for it? What version of the `indi_server` do you use?

sajmons, 2024-05-06:
I have manages to fix my building environemnt, so here is last vesrion of astra_lite 0.0.130.
In this version I did not find any errors mentioned above and I found indi_pylibcamera driver very performant. It is real pleasure to see how very fast astra_lite is compared to KStars. Live view mode and 2x2 bin is amazingly fast!

I'm closing this issue because I don't think it had anything to do with indy_pylibcamera driver.

Password for ZIP is astralite.

```
indiserver --version
2024-05-06T16:54:10: startup: indiserver --version
Usage: indiserver [options] driver [driver ...]
Purpose: server for local and remote INDI drivers
INDI Library: 2.0.7

```

[astra_lite.zip](https://github.com/scriptorron/indi_pylibcamera/files/15225350/astra_lite.zip)

closed 2024-05-06 by sajmons

scriptorron, 2024-05-07:
Thank you! I will try Astra_Lite. 

By the way: until version 2.6.0 the indi_pylibcamera driver set a wrong FITS attribute in the captured images:
the `ROWORDER`attribute was set to `BOTTOM-UP` but must be `TOP-DOWN`. That leaded to a wrong un-bayering in some post-processing tools. KStars seems to ignore this attribute and the un-bayered preview was looking good. But ASTAP and Siril (and maybe others) evaluate this attribute and interpret the Bayer data wrong. In result the images have less and wrong colors. The new driver version 2.6.2 has fixed this. It also also tries to fix an incompatibility between some versions of `numpy` and `astropy`. I hope it will still install on different Pi environments.

I have a Python script to fix the wrong `ROWORDER` attribute of FITS files. Just tell me if you need it to repair pictures you already captured.