scriptorron/indi_pylibcamera #69: IMX 296 color issue

khodgema, 2024-05-14:


scriptorron, 2024-05-15:
Sorry for this bug. I was hoping I fixed that in February :-(
Unfortunately I do not have an IMX296 camera to test. So I have not seen this bug.

Root cause are the strange exposure time limits reported by pylibcamera2 or libcamera. In your "MyCam.txt" you find the "Camera controls" with an entry ` 'ExposureTime': (29, 0, None)`. This means that minimum exposure time is 29 micro seconds and maximum exposure time is 0. I do not know if this is a bug or a feature. Anyway, for cameras reporting maximum exposure time lower than minimum I wanted to set the maximum exposure time to 1000 seconds. Unfortunately I set it to 1000 micro seconds.

I will fix this and make a new version of `indi_pylibcamera`. I will send you a message when it is done.

scriptorron, 2024-05-15:
I made version 2.6.4. You can install it with `pip`.

Please be so kind and tell me if it works.

khodgema, 2024-05-16:
The clouds rolled in before I could get focused but that seems to have done the trick, thanks for the speedy help!

scriptorron, 2024-05-16:
Do you get monochrome images and can your post-processing tools handle them? Please tell me if you see artifacts, errors or incompatibilities. I really appreciate your feedback. Thank you.

khodgema, 2024-05-16:
I'm getting color images, theirs banding and hot pixels but nothing that cant be cleaned up.

scriptorron, 2024-05-17:
Isn't it a monochrome camera? Can you send me a FITS image? I will have a look if metadata or data scaling is wrong.

khodgema, 2024-05-20:
There seems to be a color and a mono version, I have the color version. I've attached a dark with the gain maxed, I don't see anything that isn't correctable and its worked for guiding so far.

https://file.io/R0kFyhHlq95I

scriptorron, 2024-05-21:
Thank you.
The dark image looks good! There are some hot pixel but this is normal. The horizontal stripes are likely dancing between the frames. I see similar stripes with the HQ camera. Likely it comes from noise on supply or reference voltages. Until now I did not found a solution to reduce them (I tried switching off WIFI and BlueTooth, tried better supply voltage filtering, RF shielding of the whole camera, ...).
The stripes look serious due to the automatic histogram stretching of FITS viewers. But in fact the amplitudes are small and stacking removes them pretty good.

If you agree I will close this issue in the next days.

khodgema, 2024-05-22:
sounds good, thanks again for the help.

closed 2024-05-26 by scriptorron