scriptorron/indi_pylibcamera #70: warning when building

MattBlack85, 2024-08-23:
hi @scriptorron, while compiling 2.6.4 I noticed I got few warnings, are those safe to be ignored?

```
running build
running build_py
creating build
creating build/lib
creating build/lib/indi_pylibcamera
copying src/indi_pylibcamera/indi_pylibcamera_postinstall.py -> build/lib/indi_pylibcamera
copying src/indi_pylibcamera/indi_pylibcamera.py -> build/lib/indi_pylibcamera
copying src/indi_pylibcamera/make_driver_xml.py -> build/lib/indi_pylibcamera
copying src/indi_pylibcamera/CameraControl.py -> build/lib/indi_pylibcamera
copying src/indi_pylibcamera/SnoopingManager.py -> build/lib/indi_pylibcamera
copying src/indi_pylibcamera/print_camera_information.py -> build/lib/indi_pylibcamera
copying src/indi_pylibcamera/indidevice.py -> build/lib/indi_pylibcamera
copying src/indi_pylibcamera/__init__.py -> build/lib/indi_pylibcamera
running egg_info
creating src/indi_pylibcamera.egg-info
writing src/indi_pylibcamera.egg-info/PKG-INFO
writing dependency_links to src/indi_pylibcamera.egg-info/dependency_links.txt
writing entry points to src/indi_pylibcamera.egg-info/entry_points.txt
writing top-level names to src/indi_pylibcamera.egg-info/top_level.txt
writing manifest file 'src/indi_pylibcamera.egg-info/SOURCES.txt'
reading manifest file 'src/indi_pylibcamera.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE'
writing manifest file 'src/indi_pylibcamera.egg-info/SOURCES.txt'
/usr/lib/python3.12/site-packages/setuptools/command/build_py.py:207: _Warning: Package 'indi_pylibcamera.CameraInfos' is absent from the `packages` configuration.
!!

        ********************************************************************************
        ############################
        # Package would be ignored #
        ############################
        Python recognizes 'indi_pylibcamera.CameraInfos' as an importable package[^1],
        but it is absent from setuptools' `packages` configuration.

        This leads to an ambiguous overall configuration. If you want to distribute this
        package, please make sure that 'indi_pylibcamera.CameraInfos' is explicitly added
        to the `packages` configuration field.

        Alternatively, you can also rely on setuptools' discovery methods
        (for example by using `find_namespace_packages(...)`/`find_namespace:`
        instead of `find_packages(...)`/`find:`).

        You can read more about "package discovery" on setuptools documentation page:

        - https://setuptools.pypa.io/en/latest/userguide/package_discovery.html

        If you don't want 'indi_pylibcamera.CameraInfos' to be distributed and are
        already explicitly excluding 'indi_pylibcamera.CameraInfos' via
        `find_namespace_packages(...)/find_namespace` or `find_packages(...)/find`,
        you can try to use `exclude_package_data`, or `include-package-data=False` in
        combination with a more fine grained `package-data` configuration.

        You can read more about "package data files" on setuptools documentation page:

        - https://setuptools.pypa.io/en/latest/userguide/datafiles.html

        [^1]: For Python, any directory (with suitable naming) can be imported,
              even if it does not contain any `.py` files.
              On the other hand, currently there is no concept of package data
              directory, all directories are treated like packages.
        ********************************************************************************

!!
  check.warn(importable)
/usr/lib/python3.12/site-packages/setuptools/command/build_py.py:207: _Warning: Package 'indi_pylibcamera.testpattern' is absent from the `packages` configuration.
!!

        ********************************************************************************
        ############################
        # Package would be ignored #
        ############################
        Python recognizes 'indi_pylibcamera.testpattern' as an importable package[^1],
        but it is absent from setuptools' `packages` configuration.

        This leads to an ambiguous overall configuration. If you want to distribute this
        package, please make sure that 'indi_pylibcamera.testpattern' is explicitly added
        to the `packages` configuration field.

        Alternatively, you can also rely on setuptools' discovery methods
        (for example by using `find_namespace_packages(...)`/`find_namespace:`
        instead of `find_packages(...)`/`find:`).

        You can read more about "package discovery" on setuptools documentation page:

        - https://setuptools.pypa.io/en/latest/userguide/package_discovery.html

        If you don't want 'indi_pylibcamera.testpattern' to be distributed and are
        already explicitly excluding 'indi_pylibcamera.testpattern' via
        `find_namespace_packages(...)/find_namespace` or `find_packages(...)/find`,
        you can try to use `exclude_package_data`, or `include-package-data=False` in
        combination with a more fine grained `package-data` configuration.

        You can read more about "package data files" on setuptools documentation page:

        - https://setuptools.pypa.io/en/latest/userguide/datafiles.html

        [^1]: For Python, any directory (with suitable naming) can be imported,
              even if it does not contain any `.py` files.
              On the other hand, currently there is no concept of package data
              directory, all directories are treated like packages.
        ********************************************************************************

!!
  check.warn(importable)
copying src/indi_pylibcamera/indi_pylibcamera.ini -> build/lib/indi_pylibcamera
copying src/indi_pylibcamera/indi_pylibcamera.xml -> build/lib/indi_pylibcamera
creating build/lib/indi_pylibcamera/CameraInfos
copying src/indi_pylibcamera/CameraInfos/Arducam_Pivariety_IMX462.txt -> build/lib/indi_pylibcamera/CameraInfos
copying src/indi_pylibcamera/CameraInfos/IMX290.txt -> build/lib/indi_pylibcamera/CameraInfos
copying src/indi_pylibcamera/CameraInfos/Raspi_GlobalShutter_IMX296.txt -> build/lib/indi_pylibcamera/CameraInfos
copying src/indi_pylibcamera/CameraInfos/Raspi_HQ_IMX477.txt -> build/lib/indi_pylibcamera/CameraInfos
copying src/indi_pylibcamera/CameraInfos/Raspi_M3_IMX708.txt -> build/lib/indi_pylibcamera/CameraInfos
copying src/indi_pylibcamera/CameraInfos/Raspi_V1_OV5647.txt -> build/lib/indi_pylibcamera/CameraInfos
creating build/lib/indi_pylibcamera/testpattern
copying src/indi_pylibcamera/testpattern/RBG_testpattern.png -> build/lib/indi_pylibcamera/testpattern
```

scriptorron, 2024-08-24:
Hi @MattBlack85 ,

'CameraInfos' and 'testpattern' are subfolders with additional data files for development. It is not needed for operating the camera. I should move these folders to another location. For now you can ignore the warning. In a future version I will fix this.

Regards,
Ronald

MattBlack85, 2024-08-24:
Thanks for the info Ronald!

closed 2024-08-24 by MattBlack85