scriptorron/indi_pylibcamera #71: Setting to run in background?

wjschmitt, 2024-08-28:
I'm running the indiserver on a RPi4b in a virtual env and it runs fine until I close the terminal session where I started it.  Then it stops.  Is there a setting to run in the background?  Thanks.

scriptorron, 2024-09-04:
There are many option. You can also use the `screen` command which allows to get the terminal back after a disconnect. For an automated start after boot you can create a cron job with `crontab -e`  and add a line:
```
@reboot sleep 30 && $HOME/bin/YourScript.sh &
```
This will start YourScript 30 seconds after reboot. The delay allows to fully start the Pi. I am not sure if the `&` is needed at the end of the line.

scriptorron, 2024-09-19:
Solved

closed 2024-09-19 by scriptorron