scriptorron/indi_pylibcamera #75: Enhancement - Build 8bit mono FIT's file

astrogene1000, 2024-11-11:
Hi, and THANK you for providing this!

My use case is minimizing the amount of data transferred to a client from a remote indiserver.
On the scope is a PI Zero 2 W running indiserver and indi_pylibcamera and scope control.
Remote client is over wifi fetching images for the purpose of plate solving.

I am using a Pi CamV3 and configuring for RGB.
This mode builds out a FIT's file of shape(X,Y,3) 
With even a simple setting of 2304x1296 RGB and RGB Format of 576x324 (/4) give an xfer size of 567360b with headers.
Converting to mono uint8 192960b with headers, 1/3 xfer size.

Attached is a diff of CameraControl.py with my changes to accomplish this.

It is ONLY valid for Pi cameras returning a BGR888.

The as here is to extend to other Pi cameras and also add an INDI Switch to enable generating the mono frame.

Thanks!

Gene
[DiffMonoCameraControl.py.txt](https://github.com/user-attachments/files/17705541/DiffMonoCameraControl.py.txt)

scriptorron, 2024-11-12:
Hi Gene, many thanks for this great enhancement. I will implement it in the next release. But I think I will try to make it more general (also for raw Bayer images). Furthermore I will optimize speed by using numpy vector functions. Please give me some time to implement.

Regards,
Ronald

astrogene1000, 2024-11-13:
Hi Ronald,
Super!
The code 'diff' was a knock together, the 'float's killed the perf.
A quicker loop

        iq = 0
        iq1 = 0
        for i in array:
          iq = 0
          for j in i:
               sums1=int(j[0]+j[1]+j[2])
               sums1=int(sums1/3)
               if iq > 0:
                   tarr1 = np.append(tarr1,[[np.uint8(sums1)],],axis=1)
               else:
                   tarr1=np.array([[sums1],],dtype=np.uint8)
               iq = iq + 1
          if iq1 > 0:
              tarr2=np.append(tarr2,tarr1,axis=0)
          else:
              tarr2=np.array(tarr1,dtype=np.uint8)
          iq1 = iq1 + 1

BTW I did a pass using np.sum on the J0/1/2 and it was slower than the direct method above.

Also, checking performance versus network xfer time
**1) on a fast wifi link sending over the RGB was way faster versus converting to mono**
**2) On the pi zero 2w, even when framing to say 536 to 384 it went into swap which killed perf, I played with various forced garbage collections which put perf even lower**
**3) Running on PI3B 7 to 8 seconds after first frame**
2024-11-13T21:17:59: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO- got exposed frame
2024-11-13T21:18:07: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO- preparing frame as BLOB: 192960 bytes
2024-11-13T21:18:07: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO- sending BLOB
**4) On a PI5 1 second**
2024-11-13T21:10:36: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO- got exposed frame
2024-11-13T21:10:37: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO- preparing frame as BLOB: 192960 bytes
2024-11-13T21:10:37: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO- sending BLOB

So all in all, on a bad or marginal wifi link with a pi3 may pick up some total perf, on a pi zero variant it is a bust :-)

Gene

scriptorron, 2024-11-19:
I found 2 possible ways to get gray scale images fom the Image Signal Processor directely (https://github.com/raspberrypi/picamera2/issues/698): capturing YUV and taking Y channel only or capturing RGB with saturation=0. Compared to a Python conversion, both have the advantage of less memory and faster speed.

scriptorron, 2024-11-22:
I implemented monochrome captures and (hopefully correct) binning calculation in branch `monochrome_and_pixelsize_in_processed`. I still need to document the new format in the `README.md`. Furthermore I want to optimize memory usage and computing time (to make it faster on a Pi Zero) before making a release. I will do this in the next days.

scriptorron, 2024-11-26:
Monochrome captures will be implemented in next release.

closed 2024-11-26 by scriptorron

astrogene1000, 2024-11-26:
Question, not a py guy but can't some memory usage be decreased by not transposing at the start  (if doing mono)?
array = array.transpose([2, 0, 1]) # line 647

and then pulling the Y
array = array[?, ?:, ?:]  # line 665 as necessary?

Thank you again!

Gene

scriptorron, 2024-11-30:
I checked this with a memory profiler which shows me also temporary memory allocations: this line does not significantly use memory. The code looks like there is a short time where `array` and the transposed array are existing in parallel in memory. The profiler has not shown this. I think I remember that `transpose` only sets a flag in the numpy array telling that the indexes have to be exchanged when later accessing the data. This does not reorder the data itself in memory. It is really a clever implementation!

The most memory gets consumed when the
- array gets converted to FITS format,
- when the FITS gets base64 encoded and
- when a XML string with the base64 encoded FITS gets constructed for sending over network.

Unfortunately  do not see a way to make it better.

Regards,
Ronald

astrogene1000, 2024-12-20:
Hi Ronald,
Sorry for the tardy response!
Again, thank you for all of this!

I do agree there seems to be some numpy magic going on.
In my local copy I tweaked it all ways and did not see any difference in memory usage or performance.

Gene