scriptorron/indi_pylibcamera #76: FIT's header generation based on Image Settings RGB Frame Size

astrogene1000, 2024-11-18:
While working on the RGB to Mono conversion for the purposes of plate solving I ran into an issue trying to use CCDCIEL's 'Resolve' for a captured image.
The FIT's headers returned do not represent the actual values of a frame that has been downsized using RGB Frame Size.

The changes affect the following where snooped Telescope has 120mm FL RGB Frame size is set to 576x324.

**Original FIT's header's are not correct for NAXIS1/2 X/Y dimensions, pix sizes or SCALE**

NAXIS   =		       3 / number of data axes				  
NAXIS1  =		     576 / length of data axis 1				
NAXIS2  =		     324 / length of data axis 2				
NAXIS3  =		       3 / length of data axis 3				
PIXSIZE1=		     1.4 / [um] Pixel Size X, binned		         
PIXSIZE2=		     1.4 / [um] Pixel Size Y, binned		         
XBINNING=		       1 / Binning factor X				     
YBINNING=		       1 / Binning factor Y				     
XPIXSZ  =		     1.4 / [um] Pixel Size X, binned		         
YPIXSZ  =		     1.4 / [um] Pixel Size Y, binned		         
FOCALLEN=		     120 / [mm] Telescope focal length		       
APTDIA  =		    30.0 / [mm] Telescope aperture/diameter		  
SCALE   =          2.406422739 / image scale arcseconds per pixel 

**Updated FIT's header showing changed values matching X/Y dimensions**
NAXIS   =		       2 / number of data axes				  
NAXIS1  =		     576 / length of data axis 1				
NAXIS2  =		     324 / length of data axis 2				
PIXSIZE1=		    11.2 / [um] Pixel Size X, binned		         
PIXSIZE2=		    11.2 / [um] Pixel Size Y, binned		         
XBINNING=		       1 / Binning factor X				     
YBINNING=		       1 / Binning factor Y				     
XPIXSZ  =		    11.2 / [um] Pixel Size X, binned		         
YPIXSZ  =		    11.2 / [um] Pixel Size Y, binned		         
FOCALLEN=		     120 / [mm] Telescope focal length		       
APTDIA  =		    30.0 / [mm] Telescope aperture/diameter		  
SCALE   =          19.25138186 / image scale arcseconds per pixel 

[CameraControl.py_diff_of_updates_for_fits_headers_changes.txt](https://github.com/user-attachments/files/17807278/CameraControl.py_diff_of_updates_for_fits_headers_changes.txt)

scriptorron, 2024-11-19:
Hi,
I am on vacation, don't have access to my computer and can only answer based on what I remember. 
You are right, the FITS header must be adapted to the new data format. It is likely the NAXIS parameter which needs to be changed. The PIXSIZE* and *PIXSZ should not change (because pixelsize is same for RGB and monchrome).
I will work on the monochrome conversion and FITS header when I am back at home.

Regards
Ronald

astrogene1000, 2024-11-19:
Hi Ronald,
Super! And do enjoy your vacation !

Not sure I agree with not changing PIXSIZE* and *PIXSZ as the image returned, as in the case of the example of 576/324, each 'pixel' in the image covers 64X native pixels (8x8 native pixels), I think of it as software 'binning'. If the 8X binning were done by the camera the returned size would be the 1.4x8 = 11.2.

Thank you again for this indi driver!

Gene

scriptorron, 2024-11-19:
Oh, you are right with the pixel size. This is something I fully ignored in the past. My focus was on raw images. For processed images, the pixel size must be scaled by ratio of physical sensor size and software scaled image size. I will put that on my to-do list.

I enjoy the fruitful discusion with you.

scriptorron, 2024-11-22:
This should already be solved in branch `monochrome_and_pixelsize_in_processed`.

astrogene1000, 2024-11-22:
Ok! I'll give it a spin on my CamV3 and compare the fit's with one generated by my whack.
Thanks!

Gene

scriptorron, 2024-11-23:
Yes, I also need to fix the SCALE. But I believe the correct calculation is:

PixelSize(nm) * Binning / FocalLength(mm) * 0.206 = Scale(arcsec/px)

Binning is hardware (RAW frames) or software (RGB or Mono frames) binning factor.
Your calculation does not use the focal length. But it must because SCALE is the angle of the sky projected to one pixel. 

I fixed it but I could not test it. Maybe my latest commit is buggy. I will test this weekend.

astrogene1000, 2024-11-23:
Hi Ronald,

Ok, the issue, at least for me, just got a lot more complicated as it seems CCDCIEL is using more than just SCALE for it's solve_field
I find that unless BINNING=1 and PIXELSIZE1=XPIXSZ (X&Y's) it uses incorrect values to send down to solve_field

I am not asking for anything based on the above because not sure what the exact permutations are that 'fix' or 'break' it at this time!

Now back to scale
The value self.present_CameraSettings.Binning[0] does not match the calculated SoftwareBinning
Adding a log for UnitCellSize and BIN 
 self.getProp("UnitCellSize")[0] * self.present_CameraSettings.Binning[0]

2024-11-23T18:08:09: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO- UCS= 1400 BIN= 1 SCALE= (2.406425, '[arcsec/px] image scale')

Passing into snooped the SoftwareBinning =8 in example case of 576x364 focal length 120) gets me:
2024-11-23T18:16:47: Driver indi_pylibcamera: indi_pylibcamera.indidevice-INFO- USC= 1400 BIN= 8 SCALE= (19.2514, '[arcsec/px] image scale')

Thank you again!
Gene

scriptorron, 2024-11-24:
Hi Gene,

`self.present_CameraSettings.Binning[0]` is the hardware binning of the camera. It is only relevant for raw images.

I believe the calculated FITS metadata is correct. The problem is the different usage in the capture software. For instance:
- Newer versions of Ekos replace TELESCOP and FOCALLEN after receiving the camera FITS with the values from Ekos optical train. It does that without (!) recalculating SCALE. If the driver could not snoop the FOCALLEN from the mount driver (some mount drivers do not have this parameter!) than Ekos may set a FOCALLEN not fitting to the SCALE calculated by the driver. Some plate solver use FOCALLEN, others use SCALE. 
- Siril (my preferred tool for stacking raw images) seems to ignore (XY)BINNING and (XY)PIXSZ. It seems to calculate the field of view from FOCALLEN, PIXSIZE(12) (the unbinned pixel size!) and the image dimensions ignoring any binning. For binned images it is always wrong.
- CCDCiel seems to replace (XY)PIXSZ with PIXSIZE(12) and recalculates SCALE ignoring any binning. Binned pictures have now a too small field of view in the plate solver (ASTAP is used by CCDCiel). This is probably the root cause for your trouble.

The different interpretation of the FITS metadata is a nightmare. I am thinking about minimizing the metadata provided by the driver and hiding the binning used for the capture:
- PIXSIZE(12) will become the binned pixel size
- (XY)BINNING will always be 1
- (XY)PIXSZ will be removed
- SCALE will be removed
The capture software gets now images which do not seem to be binned. Only the pixel size depends on the binning set in the driver. This should work now with Ekos, Siril, CCDCiel and ASTAP. Hopefully this does not open new issues with other tools.

Regards,
Ronald

astrogene1000, 2024-11-24:
Hi Ronald,

I'll peel off the headers as you suggest and see how at least Ccdciel reacts.

Was wondering why you chose to only send RED pixels in the mono image?

# monochrome frames are a special case of RGB: exposed with saturation=0, transmitted is R channel only

Gene

scriptorron, 2024-11-24:
With `saturation=0` all 3 color channels are the same. I selected R arbitrarily. Sending channel G or B would not change something in the image. Sending all 3 channels is a waste of transmission time.

scriptorron, 2024-11-24:
I checked it right now. I made a RGB exposure with saturation=0 (driver settings -> "Camera Controls") and used FitsWork to calculate difference pictures of the 3 colors: there was only noise. This likely comes from rounding errors in the Image Signal Processor.

astrogene1000, 2024-11-24:
Hi Ronald,
Ok!

Just an FYI, to whack headers in exisiting FIT's files I use a version of modhead from FITSUTILS that allows reading a header, changing value for the header and also my mod allows renaming a header and setting value.

I have attached the 'c' source for the modified modhead.c file
Fitsutil's from NASA
[https://heasarc.gsfc.nasa.gov/docs/software/fitsio/cexamples.html
](url)
[modhead1.c.txt](https://github.com/user-attachments/files/17894127/modhead1.c.txt)

scriptorron, 2024-11-24:
Good advice! I always used `astropy` a Jupyter notebook to modify FITS header. 

I pushed the modification with the minimized FITS metadata to Git. My tests with CCDCiel look good. Maybe you can have a look on it.

astrogene1000, 2024-11-24:
Hi Ron,

I pulled the branch.
Based on the headers, CCDciel executed
solve-field --overwrite --scale-low 9.62569093024768 --scale-high 28.877072790743 --scale-units arcsecperpixp
Totally happy with the headers!

I guess "extended_Metadata" is -not- set by default(?) as I see the following headers
How would a user set extended_Metadata or is that a 'future' item?

NAXIS   =                    2 / number of array dimensions
NAXIS1  =                  576
NAXIS2  =                  324
EXPTIME =             0.999295 / [s] Total Exposure Time
FOCALLEN=                120.0 / [mm] Focal Length
APTDIA  =                 30.0 / [mm] Telescope aperture/diameter
DATE-OBS= '2024-11-24T22:35:03.884' / UTC time of observation start
DATE-END= '2024-11-24T22:35:04.884' / UTC time at end of observation
PIXSIZE1=                 11.2 / [um] Pixel Size 1
PIXSIZE2=                 11.2 / [um] Pixel Size 2
XBINNING=                    1 / Binning factor in width
YBINNING=                    1 / Binning factor in height
XPIXSZ  =                 11.2 / [um] X binned pixel size
YPIXSZ  =                 11.2 / [um] Y binned pixel size

scriptorron, 2024-11-25:
Yes, the "extended_Metadata" can be set in the INI file to get the old behavior. It is "false" by default and only a backdoor for the case that someone else complains the new behavior.

scriptorron, 2024-11-26:
I will make a new release in the next days. Than you can install it with `pip`.

scriptorron, 2024-11-30:
I released v2.7.0 now. Thank you for helping to improve this software! 

In the next clear nights I will try monochrome frames with the PHD2 guiding software. Maybe that solves my trouble with auto-guiding my mounts. In the past I always used Bayer raw images for guiding, hoping that PHD2 can manage this. But maybe PHD2 did not and failed to guide on a jagged star profile. This will not happen with monochrome images.

I will close this issue in the next days.

Regards,
Ronald

scriptorron, 2024-12-19:
No further actions needed.

closed 2024-12-19 by scriptorron