scriptorron/indi_pylibcamera #78: fits file not recognized by autostakkert

tschoenfelder, 2025-03-31:
Hello,

thank you very much for this project!

I am not sure, if this is a but or the problem lies between my ears.
Taking pictures with this device and the raspberry HQ camera and storing them as .fits ends up with files, where autostakkert will not even list them in the import window (with .fits ticked).
I observed as well, that on zoom with other .fits display tools, there seem to be black lines in the image horizontally.

Has this been observed by other or should I change some setting? I used 1x1 binning as well ...

The images taken with rpicam-still are observed by autostakkert and show no black lines, so I don't assume an issue with the camera chip ..

Best regards,
Torsten

[Polaris_Light_007.zip](https://github.com/user-attachments/files/19534343/Polaris_Light_007.zip)

scriptorron, 2025-04-01:
Hi Torsten,

I can open your file with AutoStakkert 3.1.4 on Windows 10. Please see the screen shot below. In the file dialog you must set "File type" to "image files".

[image]

I also opened your file with Gimp and Siril, zoomed in and searched for black lines. I can't find ones. The 4x4 pattern is normal when you display a raw Bayer pattern image as grayscale without debayering. This 4x4 pattern contains the color information.

[image]

Regards,
Ronald

tschoenfelder, 2025-04-01:
Hello Ronald,
thank you very much for both hints.
I downloaded your version 3.4.1 and it sees the fits files in the same folder. The 4.x version doesn't see the fits file. I took a jpg with kstars, which is seen be both. I raise it towards Autostakkert.
Thanks for the hint on debayering! As stated 'the problem lies between my ears.'.

Thank you for the time invested here and in general!
Best regards,
Torsten

closed 2025-04-01 by tschoenfelder

scriptorron, 2025-04-01:
I will download and try AS!4. I come back to you when I have more results.

scriptorron, 2025-04-02:
It seems AS!4 searches for file extension "*.fit" and ignores files with "*.fits". Your files are found when you rename them.

As far as I now there is no standard file extension for FITS files. Some tools use "*.fits", others "*.fit". It would be good when AS!4 can search for both.

tschoenfelder, 2025-04-02:
Hello Scriporron,
thank you for clearing this out! Renaming is an easy one!
Have a nice day!
Best regards, TOrsten