scriptorron/indi_pylibcamera #83: Indi_pylibcamera fails in Kstars

sc74, 2025-06-15:
Hi, I just compiled indi_pylibcamera 2.8.1 for AstroArch on RPi. But I have an issue. 

The indi_pylibcamera server works if I launch it and use it with Phd2, but if I use Kstars 3.7.6, it doesn't. 

The message in Kstars is "org.kde.kstars.ekos - "Unable to establish:\n+ INDI pylibcamera\nPlease ensure the device is connected and powered on." 
However, it finds the camera and seems to configure itself, but then it fails.

With version 2.7.0, there are no issues.

Could you tell me if I missed something during compilation or if the error is due to the data sent to Kstars by the driver?

Kstars logs:

[log_14-53-24.txt](https://github.com/user-attachments/files/20747309/log_14-53-24.txt)

scriptorron, 2025-06-16:
Hi.

I am sorry for the inconvenience. The new version fixes a bug which prevented the driver to read the default INI files from its installation directory. Now it gets the driver name from the INI file and does not use the hard coded one. In consequence the name in the EKOS setup is now "pylibcamera Main" and not "INDI pylibcamera". Alternatively you can make your own copy of the INI files and set the names as you want. Please see section "Global Configuration" in the `README.md`.

I hope this will solve your issue.

Regards,
Ronald

sc74, 2025-06-16:
Hi,

Thanks, that's perfect, I understand my mistake. Everything is working perfectly.

Best regards,
Stéphane

closed 2025-06-18 by scriptorron