scriptorron/indi_pylibcamera #91: fails to during taking a sequence with IMX290

tschoenfelder, 2025-07-30:
Dear Ronald,

taking EKOS sequence pictures, I observed several times the driver not being able to start picam:
2025-07-30T08:14:28: Driver indi_pylibcamera2: indi_pylibcamera.indidevice-INFO- sending BLOB
2025-07-30T08:14:30: Driver indi_pylibcamera2: indi_pylibcamera.indidevice-INFO- exposure settings: CameraSettings: FastExposure=False, DoRaw=True, ProcSize=(1945, 1097), RawMode={'label': '1920x1080 RGGB 12bit', 'size': (1920, 1080), 'true_size': (1920, 1080), 'camera_format': 'SRGGB12', 'bit_depth': 12, 'binning': (1, 1)}, CameraControls={'ExposureTime': 1000000, 'AnalogueGain': 1.0, 'AeEnable': False, 'AeConstraintMode': <AeConstraintModeEnum.Normal: 0>, 'AeExposureMode': <AeExposureModeEnum.Normal: 0>, 'AeMeteringMode': <AeMeteringModeEnum.CentreWeighted: 0>, 'AwbEnable': False, 'AwbMode': <AwbModeEnum.Auto: 0>, 'Brightness': 0.0, 'ColourGains': (2.0, 2.0), 'Contrast': 1.0, 'ExposureValue': 0.0, 'NoiseReductionMode': <NoiseReductionModeEnum.Off: 0>, 'Saturation': 1.0, 'Sharpness': 0.0}
2025-07-30T08:14:30: Driver indi_pylibcamera2: indi_pylibcamera.indidevice-INFO- reconfiguring camera
2025-07-30T08:14:30: Driver indi_pylibcamera2: [21:36:25.990938486] [4631]  INFO Camera camera.cpp:1205 configuring streams: (0) 192x190-BGR888 (1) 1920x1080-SRGGB16
2025-07-30T08:14:30: Driver indi_pylibcamera2: [21:36:25.994037653] [4617]  INFO RPI pisp.cpp:1483 Sensor: /base/axi/pcie@1000120000/rp1/i2c@88000/imx290@1a - Selected sensor format: 1920x1080-SRGGB12_1X12 - Selected CFE format: 1920x1080-RG16
2025-07-30T08:14:30: Driver indi_pylibcamera2: [21:36:25.999594020] [4728]  WARN IPARPI ipa_base.cpp:1392 Ctrl AeEnable is not handled.
2025-07-30T08:14:30: Driver indi_pylibcamera2: [21:36:26.049134393] [4617] ERROR V4L2 v4l2_videodevice.cpp:2049 /dev/video6[18:cap]: Failed to start streaming: Remote I/O error
2025-07-30T08:14:30: Driver indi_pylibcamera2: Exception in thread Thread-2 (__ExposureLoop):
2025-07-30T08:14:30: Driver indi_pylibcamera2: Traceback (most recent call last):
2025-07-30T08:14:30: Driver indi_pylibcamera2:   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
2025-07-30T08:14:30: Driver indi_pylibcamera2:     self.run()
2025-07-30T08:14:30: Driver indi_pylibcamera2:   File "/usr/lib/python3.11/threading.py", line 975, in run
2025-07-30T08:14:30: Driver indi_pylibcamera2:     self._target(*self._args, **self._kwargs)
2025-07-30T08:14:30: Driver indi_pylibcamera2:   File "/home/torsten/venv_indi_pylibcamera/lib/python3.11/site-packages/indi_pylibcamera/CameraControl.py", line 873, in __ExposureLoop
2025-07-30T08:14:30: Driver indi_pylibcamera2:     self.picam2.start()
2025-07-30T08:14:30: Driver indi_pylibcamera2:   File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1273, in start
2025-07-30T08:14:30: Driver indi_pylibcamera2:     self.start_()
2025-07-30T08:14:30: Driver indi_pylibcamera2:   File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1241, in start_
2025-07-30T08:14:30: Driver indi_pylibcamera2:     self.camera.start(controls)
2025-07-30T08:14:30: Driver indi_pylibcamera2: RuntimeError: Failed to start camera: Remote I/O error

[image]

Even so the timestamp doesn't really point it out, after some time I see:
2025-07-30T08:17:31: Driver indi_pylibcamera2: indi_pylibcamera.indidevice-ERROR- Uncaught exception!
2025-07-30T08:17:31: Driver indi_pylibcamera2: Traceback (most recent call last):
2025-07-30T08:17:31: Driver indi_pylibcamera2:   File "/home/torsten/venv_indi_pylibcamera/bin/indi_pylibcamera2", line 8, in <module>
2025-07-30T08:17:31: Driver indi_pylibcamera2:     sys.exit(main())
2025-07-30T08:17:31: Driver indi_pylibcamera2:              ^^^^^^
2025-07-30T08:17:31: Driver indi_pylibcamera2:   File "/home/torsten/venv_indi_pylibcamera/lib/python3.11/site-packages/indi_pylibcamera/indi_pylibcamera2.py", line 7, in main
2025-07-30T08:17:31: Driver indi_pylibcamera2:     main1(driver_instance=driver_instance)
2025-07-30T08:17:31: Driver indi_pylibcamera2:   File "/home/torsten/venv_indi_pylibcamera/lib/python3.11/site-packages/indi_pylibcamera/indi_pylibcamera.py", line 1421, in main
2025-07-30T08:17:31: Driver indi_pylibcamera2:     device.run()
2025-07-30T08:17:31: Driver indi_pylibcamera2:   File "/home/torsten/venv_indi_pylibcamera/lib/python3.11/site-packages/indi_pylibcamera/indidevice.py", line 900, in run
2025-07-30T08:17:31: Driver indi_pylibcamera2:     self.message_loop()
2025-07-30T08:17:31: Driver indi_pylibcamera2:   File "/home/torsten/venv_indi_pylibcamera/lib/python3.11/site-packages/indi_pylibcamera/indidevice.py", line 846, in message_loop
2025-07-30T08:17:31: Driver indi_pylibcamera2:     vector.set_byClient(values)
2025-07-30T08:17:31: Driver indi_pylibcamera2:   File "/home/torsten/venv_indi_pylibcamera/lib/python3.11/site-packages/indi_pylibcamera/indi_pylibcamera.py", line 176, in set_byClient
2025-07-30T08:17:31: Driver indi_pylibcamera2:     self.parent.startExposure(exposuretime=self["CCD_EXPOSURE_VALUE"].value)
2025-07-30T08:17:31: Driver indi_pylibcamera2:   File "/home/torsten/venv_indi_pylibcamera/lib/python3.11/site-packages/indi_pylibcamera/indi_pylibcamera.py", line 1409, in startExposure
2025-07-30T08:17:31: Driver indi_pylibcamera2:     self.CameraThread.startExposure(exposuretime)
2025-07-30T08:17:31: Driver indi_pylibcamera2:   File "/home/torsten/venv_indi_pylibcamera/lib/python3.11/site-packages/indi_pylibcamera/CameraControl.py", line 1031, in startExposure
2025-07-30T08:17:31: Driver indi_pylibcamera2:     raise RuntimeError("Try ro start exposure without having exposure loop running!")
2025-07-30T08:17:31: Driver indi_pylibcamera2: RuntimeError: Try ro start exposure without having exposure loop running!
2025-07-30T08:17:31: Driver indi_pylibcamera2: read EOF
<delProperty device="pylibcamera2 Guide"/>
2025-07-30T08:17:31: Driver indi_pylibcamera2: Terminated after #1 restarts.

I got the feeling, that after a timeout (60 secs or so) the restart works. I didn't stop it. The ini file is configured to the default restart.

The second camera is not active at this point of time.
Memory looks OK and no stuck process is visible.
torsten@rasppiserver3:~$ ps aux | grep camera
torsten     4542  0.0  0.0   6944  2704 pts/0    S+   10:06   0:00 /bin/bash ./start_indi_pylibcamera_short.sh
torsten     4546  0.1  0.0  96320  7968 pts/0    S+   10:06   0:01 /usr/bin/indiserver -v -p 7624 -m 1024 -r
 1 -l . indi_pylibcamera indi_pylibcamera2 indi_dewcool indi_gpsd
torsten     4547  0.1  2.1 1138944 176672 pts/0  SLl+ 10:06   0:01 /home/torsten/venv_indi_pylibcamera/bin/p
ython3 /home/torsten/venv_indi_pylibcamera/bin/indi_pylibcamera
torsten     4746  0.0  0.0   6272  1648 pts/1    S+   10:22   0:00 grep --color=auto camera

Any clue, why this is happening?

Best regards,
Torsten

scriptorron, 2025-08-01:
Hi Torsten,

the relevant error message is:
> 2025-07-30T08:14:30: Driver indi_pylibcamera2: [21:36:26.049134393] [4617] ERROR V4L2 v4l2_videodevice.cpp:2049 /dev/video6[18:cap]: Failed to start streaming: Remote I/O error

This comes from kernel driver, line 2049 in C++ file `v4l2_videodevice.cpp`. That can be anything:
- a bug in the kernel driver,
- the I/O is already in use by something else,
- the camera is not yet ready for a new exposure (but that should be handled by the kernel driver),
- communication interface to camera is configured out of specification (Do you made settings in `/boot/firmware/config.txt` for the IMX290? If so, are these settings correct?) or
- there is a hardware issue (lose contact on camera cable?)

I fear I can't do anything here.

Sorry,
Ronald

tschoenfelder, 2025-08-04:
@scriptorron 
Dear Ronald,

I am not able to reopen the IR reg. FITS header and scope.
I loaded the new driver, the config file is not using 10/10 and scope train confgured correclty, but FITS header shows focal len 0:

[image]
Is it working for you?

Best regards,
Torsten

scriptorron, 2025-08-04:
@tschoenfelder It works for me: EKOS automatically sets focal length and aperture from the optical train setup in the driver settings. My exposures have both values in the FITS header. Is it still not working for you?