scriptorron/indi_pylibcamera #97: Questions

Astro-Otter-Space, 2025-08-05:
Hello,
I just found your repository and before installing and testing your driver i have some few questions : 

1. In README, you are talking about `libcamera-apps`  but it have been replaced by `rpicam-apps` : does it still running with `rpicam-apps` ?
2. Does this driver running with [indiwebmanager](https://github.com/knro/indiwebmanager)
3. Can it be bypassed a problem i have with [compilation of indi-libcamera](https://github.com/indilib/indi-3rdparty/issues/1128)

Thank you for answers :)

scriptorron, 2025-08-05:
Hi,

regarding your questions:
1. You are right: I should replace `libcamera-apps` by `rpicam-apps` in the README. This is a relict from the time when I started that project. The driver does not use `libcamera-apps` or `rpicam-apps`, it directly communicates with the OS. To answer your 1st question: yes it works.
2. I always start the INDI server with a self made shell script. I newer tried `indiwebmanager`. So I can only guess: The `indi_pylibcamera` driver is not part of the standard INDI library. Likely `indiwebmanager` will not find the driver without additional tweaking.
3. For using the `indi_pylibcamera` you do not need to compile the INDI driver. I use the INDI packages coming with the Raspberry Pi OS. 

Regards,
Ronald

Astro-Otter-Space, 2025-08-06:
Thank you for answers.
I've installed your driver and if i understand correctly (english isn't my native language ^^) i need to execute indiserver in python virtual environment, right ? It can't be running as systemctl service (like i do with indiwebmanager) ?

I don't see `indi_pylibcamera.xm` file in `/usr/share/indi` : is it a bug ?